bambulab/BambuStudio #2679: says to home when command to move bed up or down after completing a print? but moves it anyway?

thetechy2, 2023-10-07:
### Bambu Studio Version

1.7.5.92

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Additional system information

shouldnt matter but gigabyte z390 ud, i5 9600k with rtx 2070 and 32gb ddr4

### Printer

x1c with ams

### How to reproduce

1 do any print
2 let it finish
3 command move build plate up or down from bambu studio

### Actual results

it shows the dialouge box to home before moving z, but moves z anyway, and will force you to close the dialouge box window to move the bed another 10mm

### Expected results

just move the bed without the dialouge box 

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-10-12:
Hi. @thetechy2 
This is actually a form of protection, and moving the hot bed is not a high-frequency operation. We can try adding an option to turn these prompts on or off,

walterwongbbl, 2023-12-21:
If you have any other issues, please create a new issue. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl