bambulab/BambuStudio #2680: Axes do not display correctly when flipped to different sides. to use scaling

Witawat, 2023-10-08:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1C Combo

### How to reproduce

Limitations and confusion in Set the desired axis scale.

[image]

### Actual results

In Prusa X Y Z Axes
[image]

When flipped over, the axes display correctly.

[image]

In Bambu
[image]

When flipped to different sides, different axes are displayed along the workpiece. There is a problem with confusion in scaling various workpieces and axes.

[image]

### Expected results

Want to display various axes constant, Prusa style, not changing along the workpiece.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12840368/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-10-11:
​There are many changes to fix this BUG. We will improve it in the future.

Witawat, 2023-12-19:
> We will fix it in the future.

wait long time?

OzDruiD, 2023-12-20:
@zhimin-zeng-bambulab Its not just in flipping \ changing , the raw values are swapped.
Look at the X\Y here - clearly the "Y" length of the boat is 60, and the "X" width of the boat is 31  - but thats not what the Scale screen says.  
[image]
More reports on Bambu Forums
https://forum.bambulab.com/t/x-and-y-colour-do-not-match/43220

Witawat, 2023-12-30:
@zhimin-zeng-bambulab 
How long will it take to fix errors?
I think it's urgent. Because it involves parallel adjustment.
Because bambu doesn't allow us to adjust the estep value ourselves.

zhimin-zeng-bambulab, 2024-10-21:
[image]

It has been fixed in  1.10 Beta1.

closed 2024-10-21 by zhimin-zeng-bambulab