bambulab/BambuStudio #2681: Bambu Studio Crash on import or export 3mf

bumpusd, 2023-10-08:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home 22H2 19045.3448

### Additional system information

Processor	Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz   2.81 GHz
Installed RAM	16.0 GB (15.9 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Babmu Lab P1P

### How to reproduce

1. Attempt to import any 3mf file (from local storage or MakerWorld)
2. Attempt to export any 3mf file

### Actual results

1. Bambu Studio loads the file and before you can do anything the software closes
2. Bambu Studio closes as soon as export completes

### Expected results

Bambu studio should allow you to import and export 3mf files (even ones from outside sources) without closing the software.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12841019/log.zip)
[custom supports Mod_Paint_Rack_Tamiya_33oz.zip](https://github.com/bambulab/BambuStudio/files/12841022/custom.supports.Mod_Paint_Rack_Tamiya_33oz.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-10-09:
@DanBao-Bambu 帮忙分析一下，有 DMP

DanBao-Bambu, 2024-01-17:
This issue may be due to you setting "Maximum recent projects" to 0. You can use the latest version 1.8 to see if this issue has been resolved. If it has not been resolved, please provide feedback to us. This may be another issue.