bambulab/BambuStudio #2683: "repair model" should not only be on"windows"BLStudio,  that should have been available on "MAC" or any other version for that matter

simplymeandi, 2023-10-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

QingZhangBambu, 2023-10-09:
#2501

QingZhangBambu, 2023-10-09:
#183

QingZhangBambu, 2023-10-09:
[image]
you can check #183 about that