bambulab/BambuStudio #2689: PA Profile Factor K not shown in German language

MrTom1, 2023-10-10:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Ventura & Sonoma 

### Printer

Bambu Lab X1C

### How to reproduce

Under device - AMS settings, 'Factors of Flow Dynamics Calibration', 'Factor K' is always set to 0,000000 in German language, however works correct in settings for English language

### Actual results

Factor K not useable in German language

### Expected results

Factor K should be set-up as in English language settings

### Project file & Debug log uploads

[debug_Tue_Oct_10_11_39_24_13672.log.0.zip](https://github.com/bambulab/BambuStudio/files/12855720/debug_Tue_Oct_10_11_39_24_13672.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-10-12:
[image]

Can you see your calibration results under Calibration page in a German environment?

MrTom1, 2023-10-12:
[image]

Results are all set to 0,000!

MrTom1, 2023-10-12:
Another bug on this:
When I'm in the 'Manage Results' window (as above) and try to 'Edit' the result of filament, I can do it and the changed K Factor is shown in the table. However than I'm stuck on this window, no further 'Edit' or 'Delete' of any filament are possible.
Closing the window and reopening 'Manage Results' are showing 'Faktor K' values back to zero, changes done before are gone.

zhimin-zeng-bambulab, 2023-10-13:
It has been fixed in V1.7.7.89, you can have a try.

MrTom1, 2023-10-13:
Confirmed, working fine now!. Bug fixed.
Thanks

closed 2023-10-13 by MrTom1