bambulab/BambuStudio #2694: Timelapse/Video recording plays in less than 1/2 second

Waffull, 2023-10-11:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Not relavent

### Printer

X1C

### How to reproduce

1: Download recording
2: Play with VLC, Windows Media Player, anything that plays mp4 videos
3: Entire video plays in about 1/2 to 2 seconds...  regardless of length of print (even tried with a print that took 8 hours.)

### Actual results

 Entire video plays in about 1/2 to 2 seconds...  regardless of length of print (even tried with a print that took 8 hours.)

### Expected results

A reasonable time lapse, depending on length of print.  Should be able to see each layer and parts of each layer, as well.

There doesn't appear to be any settings to configure video recording.  Either on or off and 720 or 1080.

In earlier versions of Bambu Studio, this worked.  I haven't played with it for the past 3 or 4 versions, so not sure when it changed/broke.

### Project file & Debug log uploads

https://github.com/bambulab/BambuStudio/assets/12578789/4842d56b-072f-4880-ac7e-ef752acfc4c0

This was a 7 hour print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-10-11:
@Waffull Is the video recording file in the printer SD normal?

Waffull, 2023-10-13:
Yes, the videos files, not the timelapse files, look fine.  They are roughly 250mb each.

bambu123, 2023-10-16:
> Is the video recording file in the printer SD card normal?

Sorry. I want to ask if the timelapse file in the printer SD is normal. If it is also a very short mp4 file, then it should be a problem with the printer generating timelapse. I need your help to submit the ticket.