bambulab/BambuStudio #2695: Same issue here, Slot 1 and 3 have exactly the same:

hly1993, 2023-10-11:
Same issue here, Slot 1 and 3 have exactly the same:
- Color
- Filament Type
- Profile on Bambu Studio

Forced a resync of the AMS in BambuStudio

Still get the message: "There are currently no identical spare consumables available [...]"

[image]

Really surprising to see such a simple featuring being this unreliable.

_Originally posted by @GuillaumeLeclerc in https://github.com/bambulab/BambuStudio/issues/2103#issuecomment-1738151879_

lanewei120, 2023-10-11:
> ```
>           Same issue here, Slot 1 and 3 have exactly the same:
> ```
> 
> * Color
> * Filament Type
> * Profile on Bambu Studio
> 
> Forced a resync of the AMS in BambuStudio
> 
> Still get the message: "There are currently no identical spare consumables available [...]"
> 
> [image]
> 
> Really surprising to see such a simple featuring being this unreliable.
> 
> _Originally posted by @GuillaumeLeclerc in [#2103 (comment)](https://github.com/bambulab/BambuStudio/issues/2103#issuecomment-1738151879)_

hello
@hly1993 which machine are you using?
currently P1P and A1 mini don't support this feature, and we will add it soon

could you help to upload the logs for us to check the details

walterwongbbl, 2023-10-12:
Hi. @hly1993 
If you enable filaments backup in AMS settings, filaments backup will be automatically used, but printers of the p1p and a1 series are currently unable to push a list of mapping relationships between filaments.
After the printer can push the correct replacement relationship in the future, a diagram of filament replacement will be displayed here.

walterwongbbl, 2023-12-21:
If you have any other issues, please create a new issue. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl