bambulab/BambuStudio #2697: Adding folders to store project files on the printer

DronBroks, 2023-10-11:
**Is your feature request related to a problem? Please describe.**
I use Bambu Lab X1C for work. For printing, I have about 500 different models, which have certain categories and are located strictly in their own folders on the computer. Since downloading files before printing is only possible from a desktop computer (it is hampered by the poor performance of the cloud; it is often impossible to download files to the printer and you have to copy the files directly to the card), I am forced to keep almost all working files downloaded to the printer. Most often, printing is launched through a mobile application. And here the most terrible situation arises when you need to find the desired file among 300-400 files located in one list.

**Describe the solution you'd like**
I believe that the ability to sort project files into folders (directories) on the printer will make working with files much easier not only for me but also for other users of your equipment. Previously, I read requests to add the ability to work with external folders located on external servers, which also stems from the problem of unsatisfactory operation of the system’s cloud server and difficulty working with a single list of files in the printer’s memory.

DronBroks, 2023-10-27:
I see there is no reaction from the developers to my question. Then we recommend visiting the forum and reading about what users are interested in having a file system in the printer’s memory.

DronBroks, 2023-11-09:
?

walterwongbbl, 2023-11-09:
Hi @DronBroks 
We will discuss this requirement

DronBroks, 2024-01-11:
Interesting, do you at least sometimes look at your forum, for example, this topic https://forum.bambulab.com/t/add-folder-support-to-sd-card/3790
People have been discussing this issue in it since January 2023. Provided that such a scheme for constructing a section with files for printing is organized even on printers for 200 bucks, but on a printer for almost 2000 bucks we see the beginning of the 20th century. I'm just tired of poking around and looking for the file I need to print among 300 files.

CoReYeDe, 2024-01-15:
+1 
I have so many files that i print recurring and my sd-card is full with different patterns of filenames to get it sorted..
Foldersupport would be a great improvement to handle the files. <3

chilternflyer, 2024-01-16:
This is basic stuff that every other printer has. I don't understand why it's not in the P1 series firmware already.

antonvafsp, 2024-01-24:
I need this feature! I have up to 1000 gcode files on my very old printer, running very old firmware. And now, in 2024, i don`t have this feature on NEW printer!!!

Xtreemtec, 2024-02-20:
We use the Bambulab printer as our part printer for many products we build in our office.   We have several 100 parts that we need to re-print often and now we are exporting them over and over again from Fusion360 as that is faster then starting to look for the part on the printer.

Just basic 8.3 FAT filename support would allow us to group the files into folders. with the product name to the folder.  

Would not even mind to hand sort the files on the SD card.    As long as Folder support is there. ;)

sodachen22, 2024-02-21:
Thanks for the suggestion, we are evaluating the solution to support fold on SD card. :)

mackbon401, 2024-03-22:
+1 for this! I run a university makerspace and we have 15 XC-1 printers on their way. There are hundreds of students that come through the lab to print things. I'm really hoping this gets resolved by August when we officially open our new space, otherwise it is going to be hectic.

Even our crappy Lulzbot printers have folders!

DronBroks, 2024-04-11:
Exactly six months have passed since the issue was raised. There is zero information from the developers. It’s even hard to imagine what causes such a problem. If even on cheap printers there is support for folders with files for sorting them.

DronBroks, 2024-05-14:
The discussion among the developers dragged on for a long time. Apparently they are developing the file system from the very beginning. Meanwhile, the profile topic on the forum is filled with requests to do in the printer what even budget companies did 8 years ago.
Whatever you are looking for, here is a link to the profile topic
 https://forum.bambulab.com/t/add-folder-support-to-sd-card/3790

HStonH, 2024-06-12:
Very much needed here too!

HStonH, 2024-06-13:
It's such a pain to scroll through 300+ Files without any sorting or folder structure... madness

RyanEwen, 2024-06-27:
It's extremely frustrating to print projects from SD without any way of organizing them into folders or sorting or filtering.

iscarelli, 2024-07-28:
No way. More than a year to have a way to use folders in SD card?

DronBroks, 2024-07-29:
It seems the developers don't read this topic at all. They don't care about our legal demands to change the way we display project information.

cederron, 2024-09-25:
we need this

NevesLF, 2024-10-24:
I'm honestly considering switching back to Sovol because of this. The A1 is an amazing printer, but this is such a dealbreaker that it's making it almost unusable to me.

It's also such a basic feature that I didn't even consider it might not be available before buying the printer.

DronBroks, 2024-10-30:
Yes, they do not care about our wishes. This request has been hanging for a year and more than 10 months have passed since the developers wrote what they would think about this issue. So who needs a file system in projects, apparently you need to look for another manufacturer of printers. This manufacturer was stuck in the 20th century at the dawn of the formation of DOS systems.

johnnydoesdesign, 2024-11-21:
+1 Just started using a bambulab P1S - but ive already lost myself in clicking through the SD card.
PLEASE! ... please add this. 
And if not directly on card, let me at least have folders in bambustudio view of the stored files so i can start the print from my computer (maybe done with tags, and leaving the files in the root directory).

bmizes, 2024-11-29:
Just bought a p1s and I can't believe this feature is missing.  I wouldn't have bought the printer if I had known about this. This is a BASIC feature, not some advanced concept.  Folders have been around since the dinosaurs.  Bambu printers are supposed to be user friendly but having a gigantic list of files on an sd card makes reprinting almost impossible.  Time to get this done yesterday!!!

Thr33phase, 2024-12-01:
My Creality's and Prusas have supported folders for years.   Wondering when this will be available.  It is a limiting factor in switching our 100+ printers to Bambu without changing our entire process.

jryan15, 2024-12-30:
Please add this incredibly basic feature!

claytonhoxworth, 2025-04-01:
Still waiting on this feature (for more than 2 years now). BambuLab devs, please add it. This is a file structure any freshman in college could implement.
I really hope they are not refusing to add it so people are forced to use their cloud service to transfer sliced files. This system would not draw people away from your service; it would just make print farm users (like myself) have a much more pleasant and less labor-intensive time starting prints. Bambu, please have an intern or someone take half of their shift to make this change. This is the most annoying thing about your machines.

RyanEwen, 2025-04-01:
The lack of this feature and other basic features like it have caused me to start looking at machines like the Elegoo Centauri Carbon and the Anycubic S1 instead of just getting more P series machines. 

I feel like Bambu has already forgotten about many of its users.

Xtreemtec, 2025-04-02:
> I guess Bambu Labs does not want SD Card printing to be too comfortable. This feature would take a couple hours of coding to implement, but they prefer your prints going through the cloud.

I do not agree with you.  They also implemented a Direct mode in the Bambu Software to send your files straight over the network without cloud.  
Also files are now stored on the SD card after sending them to the printer..  But being unable to sort these files into categories the list grows longer and longer..

We have this printer in our company and besides the tooling and prototyping.   We use the printer very much for our production,  Creating 3D parts that we need in our product housings to hold a Fan, a display etc..

It must be as easy for our employees to walk up to the printer.   Hit the SD card, Select a folder of the product, and then hit the 3D part they need.  ( Often plates with 10/20 pieces at once ) 

Not being able to sort and put them in sub folders is just frustrating as they keep on searching trough files..  Once in a while even print the wrong file.. 

So while we are very happy with the printing speed, reliability, and quality of the printer and it's 3D parts..   We really considering to not purchase the H2D and look further to other brands that DO support folder structure n the SD card for easy reprint of parts already in the machine.

GiveWave, 2025-05-02:
Come on bambu y'all just released a new machine that no one asked for but this request that's been around for at least 2-3 years from what I can tell STILL hasn't been sorted with an update !? ... I bought 3 X1s to replace my my old ass prusa mk3s, and as much as I do love my new machines I sure as as hell would've gone with a different company if I knew this basic feature wasn't available as I manage hundreds of files. Having the ability to have a folder structure would save so much wasted time instead of endlessly scrolling for the ONE file needed every time. PLEASE add this already !!! Pretty sure y'all are the only company in 2025 that does not have this basic feature that even the very first maker bots did -____-

claytonhoxworth, 2025-07-04:
I’m genuinely disappointed. The machines themselves are excellent, but the software, especially the difficulty of navigating an SD card full of files, makes it feel as if you want everything funneled through your cloud service/application. BambuLab's silence on adding this simple feature just reassures this notion. Why has there been no progress on simplifying SD card printing for users? It almost seems intentional, as if you’re forcing us to use your app by making offline printing a struggle. And by adding further obstacles to working with other slicers, you only reinforce that impression. Again, the implementation of this is super easy. But the lack of the addition of this feature in software updates (and new machines like the H2D) is very disappointing.

Please, BambuLab, for the sake of customer trust, could you at least acknowledge this request?

RyanEwen, 2025-07-04:
> Well. It’s not that you can not print without there cloud service.  There is a Lan only printing mode. There is a way to see which models are on the sd card.  It just lacking a Fat 8.3 file system which is even on small 8bit embedded processors easy to integrate.  But the lack of this functionality on the Bambu printers really made me not upgrade to the H2D..We really wanted this IF it had a working file system for at least 1 level of sub folders. That at least gives us the possibility to order and structure the files on the card and have an easy way to reprint a lot of the parts we make make on our printer.  Our Bambu is used 3-4 days a week to run parts for several products we make in our company.   Without needing to scroll a lot or to upload the file again and again from studio. Also WHY after 2 years we still have the starter models which I can’t seem to get removed.  Don’t need them but they take up space on the screen every time…. Bambu start listening to your users.  Because you make one of the finest printers out there.  With very advanced camera detection and AI stuff.  But something simple like a Fat filesystem which dates back to DOS in the 70’s is too hard? Daniel WittenaarOp 4 jul 2025 om 20:04 heeft Ryan Ewen ***@***.***> het volgende geschreven:RyanEwen left a comment ([bambulab/BambuStudio#2697](https://github.com/bambulab/BambuStudio/issues/2697))
> I also feel like it's being done on purpose. You can't do anything useful with an attached AMS using the built-in screen on a P-series machine either. Competitor machines let you do more for less.

The file system does support folders and you can even mange them using FTPS (I am the one with the giant FTP thread in their forum). They just don't have support in the UI to show the folder structure. If you make it yourself over FTPS then the printer UI shows the contents of all the subfolders as if they were all together in the parent folder

Xtreemtec, 2025-07-04:
What is the point of having folders if the system can not show it? I mean yes I have made some folders to try it.  But you want to go to the sd card and see a few folders you made to organise the many files.  My elegoo resin printers have it.  Even my 8+ years old flashforge had folder structure.  It doesn’t help that everything is 1 big heap of files. Met vriendelijke groet,Daniel Wittenaar

RyanEwen, 2025-07-04:
> What is the point of having folders if the system can not show it? I mean yes I have made some folders to try it.  But you want to go to the sd card and see a few folders you made to organise the [...]

I am in agreement with you. You said the file system doesn't support folders, but I am saying that this is even sillier than that. The filesystem does support subfolders, and they already have logic to show files from subfolders, yet they limit the UI to displaying only a flat list. It's ridiculous.

GLOUPY-RnD, 2026-02-26:
I'd like to be able to manage folders ! Printing projects with 200pieces each is a pain currently...

tropho23, 2026-03-22:
@walterwongbbl or @sodachen22: it's been three years and thousands of users across the internet have requested this feature, and there has been no response from the developer team.  Even a "it's hard, we're working on it" would be nice.  My much cheaper Creality K1C printers have this feature, as do many other printers with a physical screen UI.

Please reply, we really need this feature.