bambulab/BambuStudio #2699: The feed button is broken

floppyhammer, 2023-10-11:
### Bambu Studio Version

V1.7.6 & V1.7.7

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Clicking the feed button in the studio has no effect. Every time I have to initiate the feeding op on the printer. No problem with unfeeding though.

### Actual results

The feed button doesn't work.

### Expected results

The feed button should work as it did in previous versions.

### Project file & Debug log uploads

No need.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-10-12:
Hi. @floppyhammer 
May I ask how you operate it? You need to first click on the AMS tray you want to manipulate, and then click on the feed button

floppyhammer, 2023-10-12:
@walterwongbbl I don't have the AMS installed.

[image]

walterwongbbl, 2023-10-12:
Hi. @floppyhammer 
You need to first select the ext spool and then click the load button.
If you click on load directly, you don't know which track to load.

floppyhammer, 2023-10-12:
@walterwongbbl Yeah it works. I wonder if this step is necessary in previous versions.

closed 2023-10-13 by floppyhammer