bambulab/BambuStudio #2702: error message that the temperature is to high to start print everytime i try to print.

Thehaus64, 2023-10-11:
### Bambu Studio Version

the newest one

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

Bambu X1

### How to reproduce

just try to print and the error pops up. my temp was 30 and the print needs a heated bed of 55 for pla so i should not receive the error at all let alone every time i try to print. it started as soon as i updated to the new bambu studio. 

### Actual results

it pops up and says that the temp is to high do i want to proceed

### Expected results

it should just print no error message

### Project file & Debug log uploads

hg lu

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2023-10-16:
Sorry,the tip should be "The hot bed temperature in current setting..." rather than "The current hot bed temperature". In latest version , we remove the pop up and add a daily tip for open-door printing

closed 2023-11-14 by QingZhangBambu

dsperman, 2024-01-16:
I have the same problem.  I am on firmware 1.07.02.00 and I get the message every time I print PLA with the bed temp at 55 (which is every time I print PLA.)