bambulab/BambuStudio #2704: When naming a plate the Plate Name text box should automatically be in focus

the-robo, 2023-10-11:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Not a performance issue.

### Printer

X1C AMS

### How to reproduce

1. Click the pencil icon to name a plate
2. The Plate Name text box should be in focus, however it is not. The user must then click the text box to enter the plate name.

### Actual results

The Plate Name text box should be in focus, however it is not. The user must then click the text box to enter the plate name.

### Expected results

The Plate Name text box should be in focus after the user clicks the Pencil icon to name a plate.

This becomes more irritating as additional plates are added to the project. In one project I added 20 plates and was quite frustrated at having the extra click just to put the plate name textbox in focus.

### Project file & Debug log uploads

This is a design issue and does not require log files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2023-10-12:
[image]
Hello, the issue with the plate name cursor has been resolved. Please look forward to version 1.8

Haidiye00, 2024-01-16:
The issue has been fixed, please use the latest version V1.8

closed 2024-01-16 by Haidiye00