bambulab/BambuStudio #2706: Couldn’t connect to server

Rainer-prog, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

macOS: Ventura 13.6

### Printer

BambuLab X1-C

### How to reproduce

1.Go to "device"

### Actual results

Result: No printer

### Expected results

Start printing process

Error: 
Thu Oct 12 16:32:36:test bambulab start…
Thu Oct 12 16:32:36:test bambulab failed
Thu Oct 12 16:32:36:status=0, body=, error=curl:Couldn’t connect to server:
Failed to connect to api.bambulab.com port 443: Bad file descriptor
[Error 7]

### Project file & Debug log uploads

[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/12883724/BambuStudio.zip)
project-file bigger-spool-inserts-v3.3mf.zip

### Checklist of files to include

- [X] Log file
- [X] Project file

Rainer-prog, 2023-10-12:
Sorry. It's working again. I forgot Little Snitch which stopped outgoing traffic.

QingZhangBambu, 2023-10-13:
That's fine, I am glad that your issue has been solved.

closed 2023-10-13 by QingZhangBambu