bambulab/BambuStudio #2707: Can't print generic filaments but only BL

DirkFlorin, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.0

### Additional system information

CPU/GPU: Apple M2 Max
Memory: 64GB unified memory
Harddrive: 1TB ssd
LAN: 10GbE

### Printer

Bambu Lab X1C

### How to reproduce

Insert any other brand of filament but Bambu Labs filaments in the AMS and try to print something
I tried it with PLA and I can't use any generic ones - yes, its flagged as "PLA"

### Actual results

"Note: Only the AMS slots loaded with the same material type can be selected."

### Expected results

Using generic PLA and print, just like before. With version 1.7.6 its working fine for me. No issues, no nothing. But its buggy with Version 1.7.7

### Project file & Debug log uploads

[02.TopCase.3mf.zip](https://github.com/bambulab/BambuStudio/files/12884382/02.TopCase.3mf.zip)
[debug_Thu_Oct_12_17_22_07_83732.log.0.zip](https://github.com/bambulab/BambuStudio/files/12884383/debug_Thu_Oct_12_17_22_07_83732.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

StoneLiBambulab, 2023-10-18:
Hi @DirkFlorin 
you can select Generic PLA when using other brands, the AMS filaments settings are only related to AMS mapping and filament switching.

Thanks!
Stone Li（旺仔）

DirkFlorin, 2023-10-19:
Hi @StoneLiBambulab !

Thank you for your reply, but that's what I did and it does not help. But I have found the issue, that one of the filaments profiles was set to "PLA+" in PLA type. It was not selected and it was not in the AMS. I wasn't able to choose any PLA, only BL PLA was available to chose. The "bug" is still present, but it was working fine in 1.7.6. So it seems like, you changed something where the Slicer is checking the filament presets, whether they are selected or not.

DirkFlorin, 2023-10-25:
Hi @StoneLiBambulab 

There is really something broken since version 1.7.7 first released with that the AMS is sending to your cloud or how the slicer is handling with that data. I put a spool of Prusament PETG in the AMS and also told the AMS that its Generic PETG. I sliced it with BL Studio and now when I try to print it, it always tells me "Please check each filament above to specify its mapping AMS slot before sending the print job" and I am not able to select the slot. But in BL Studio it says "Generic PETG in Slot 1".

With version 1.7.6 its working flawless without any problem.

So there must have been a change in the code.

Kind regards,
Dirk

StoneLiBambulab, 2023-11-06:
There is a bug when getting the filament type, will be fixed in 1.8.xx

Thanks!
Stone Li

DirkFlorin, 2023-11-10:
Hey @StoneLiBambulab !

I can confirm, that the issue has been fixed with the current Beta of 1.8.X of Bambu Studio. Thank you again for getting back to me and thanks also for fixing the issue.

You guys do an amazing job and provide a great software.

Kind regards,
Dirk

closed 2023-11-10 by DirkFlorin