bambulab/BambuStudio #2708: Filament issue on Bambu Lab

choseverte, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

MacOs 13.5.2

### Additional system information

Just before the issue, I installed the last firmware ( I had no choice, it was pushed by the application in order to print)

First error messsage was HMS_WARN_07FF_2000_0002_0004

On the printer screen, the AMS is shown as active but nothing happens and on the application, the filament appears on the external feeder (even if it is empty and never used)

### Printer

Bambu Lab P1S (with 1 AMS)

### How to reproduce

Lanch any printing

(I uploaded the last I tried but any project will come with the same result, it looks more like an issu of the filament detector in the printing head and probably something with the AMS)

### Actual results

[image]
The printer imagine a filament is blocked from external spool bur first I never used it, second I totally disamount and cleaned everything, this is not the case.

Additionally, since it happens, my AMS is not recognising any filament, does not start anything.
The printer is blocked as shown on the printscreen.

In teh AMS, the second filament is in (about 1 inch) but nothing happens)

I already tryed to check the head, in the doubt I even changed the nozzle. 

I have the impressiont that 1st the AMS has an issue, 2nd, the little sensor (magnetic one) is not anymore mobile and therefore, the system think it is filled with an external filament.
Accordind ti the wiki, I am supposed to contact customer support

### Expected results

I need help in order to restart the printing process

### Project file & Debug log uploads

[Kabelhalter3MF.3mf.zip](https://github.com/bambulab/BambuStudio/files/12885729/Kabelhalter3MF.3mf.zip)
[debug_network_Thu_Oct_12_18_50_09.log](https://github.com/bambulab/BambuStudio/files/12885826/debug_network_Thu_Oct_12_18_50_09.log)
[debug_network_Thu_Oct_12_20_07_01.log](https://github.com/bambulab/BambuStudio/files/12885827/debug_network_Thu_Oct_12_20_07_01.log)
[debug_network_Wed_Oct_11_20_17_24.log](https://github.com/bambulab/BambuStudio/files/12885828/debug_network_Wed_Oct_11_20_17_24.log)
[debug_network_Wed_Oct_11_20_39_54.log](https://github.com/bambulab/BambuStudio/files/12885829/debug_network_Wed_Oct_11_20_39_54.log)
[debug_network_Wed_Oct_11_21_33_24.log](https://github.com/bambulab/BambuStudio/files/12885830/debug_network_Wed_Oct_11_21_33_24.log)
[debug_network_Wed_Oct_11_21_40_02.log](https://github.com/bambulab/BambuStudio/files/12885831/debug_network_Wed_Oct_11_21_40_02.log)
[debug_network_Wed_Oct_11_21_40_57.log](https://github.com/bambulab/BambuStudio/files/12885832/debug_network_Wed_Oct_11_21_40_57.log)
[debug_network_Wed_Oct_11_21_45_27.log](https://github.com/bambulab/BambuStudio/files/12885833/debug_network_Wed_Oct_11_21_45_27.log)
[debug_network_Wed_Oct_11_22_07_20.log](https://github.com/bambulab/BambuStudio/files/12885834/debug_network_Wed_Oct_11_22_07_20.log)
[debug_network_Wed_Oct_11_22_23_42.log](https://github.com/bambulab/BambuStudio/files/12885835/debug_network_Wed_Oct_11_22_23_42.log)
[debug_network_Wed_Oct_11_22_23_50.log](https://github.com/bambulab/BambuStudio/files/12885836/debug_network_Wed_Oct_11_22_23_50.log)
[debug_network_Wed_Oct_11_22_44_02.log](https://github.com/bambulab/BambuStudio/files/12885837/debug_network_Wed_Oct_11_22_44_02.log)
[debug_network_Wed_Oct_11_23_17_35.log](https://github.com/bambulab/BambuStudio/files/12885838/debug_network_Wed_Oct_11_23_17_35.log)
[debug_network_Wed_Oct_11_23_22_47.log](https://github.com/bambulab/BambuStudio/files/12885839/debug_network_Wed_Oct_11_23_22_47.log)
[debug_network_Wed_Oct_11_23_42_15.log](https://github.com/bambulab/BambuStudio/files/12885840/debug_network_Wed_Oct_11_23_42_15.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

choseverte, 2023-10-21:
Problem solved! After checking all the different elements, the magnetic sensor of the feeder seems to be broken.
Disasembled everything, noticed that the little spring was broken, order new parts, chend it and now the printer is working again!

jcepollina2k, 2025-05-16:
how did you remove enough of the feeder to get to the magnetic sensor? any youtube or other video links? i've hit a similar issue and just cleaned out extruder but now ams can't detect any filament and still thinks there is filament stuck in hot end.