bambulab/BambuStudio #2710: Flush options, Flush into objects' infill: disabled but still happening

RussMathis, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

intel core 9: 16G

### Printer

Bambu Carbon x1

### How to reproduce

create simple model requiring support, add text to it, set model to white play, set text color to black.
Enable support, I use tree style thin.
Disable flush into objects' infill
Disable flue into objects' supports
---slice plate
Notice that the black is flushed into supports.
Also notice, while the preview does not show black flushed into the object's infill, when printing black HAS been flushed into object's infill, even though it has been disabled.

### Actual results

Flushing not working properly. Even when disabled flushing takes place.

### Expected results

As described by documentation.

### Project file & Debug log uploads

[ProjectWithFlushingIssues.zip](https://github.com/bambulab/BambuStudio/files/12886637/ProjectWithFlushingIssues.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-10-13:
[image]
Because your supports and support interfaces use the default material, this will cause the supports to directly use the material currently being used.  If you want to specify the material of the support, you can set the support material.

zhimin-zeng-bambulab, 2023-10-17:
@RussMathis Do the above suggestions solve your questions?

zhimin-zeng-bambulab, 2023-11-13:
This issue is not a bug, close as no response.

closed 2023-11-13 by zhimin-zeng-bambulab

gpeden, 2024-03-20:
also check that prime tower is enabled in the Others tab.  If its not, the flushing options per object are grayed out.  It is not obvious that these are related.