bambulab/BambuStudio #2711: [Issue] Studio asks at every startafter  SSl certificates

BambuLabDiscord, 2023-10-12:
The message in starting dialog should only displayed once. 
Seen on Archlinux with KDE

bambu123, 2023-10-13:
Not sure what you are talking about. Could you please take a screenshot of the dialog?

tkrill, 2023-10-24:
I most likely also experiencing this while running Ubuntu 22.04. This shows up on every start of Bambu Studio. If i choose No in the dialog the application terminates. (I'm also totally unable login from the app when launched which might be related or not) 

[image]

justpie, 2023-12-22:
I am also getting this. Any luck resolving it?

tkrill, 2023-12-23:
@justpie i managed to login and then i the problem stopped. Unfortunately i don't remember if i did something differently.

lanewei120, 2023-12-25:
Hello @tkrill 

if click "remmember my choice"
will it still popup when bootup next time?

justpie, 2023-12-25:
I actually was able to resolve my issue by clicking "remember my choice" and logging in. However, there was a GUI issue where the checkbox label wasn't visible. All I could see was the checkbox so it made it really unclear as to what I was actually clicking on.

tkrill, 2023-12-25:
@lanewei120 i unfortunately don't remember but an educated guess is that i checked the remember box. (If there is a way to reverse this i can try that. Otherwise @justpie  comment might be enough)

lanewei120, 2023-12-26:
thanks

we will fix the black  checkbox issue, make it more clear

pcairic, 2024-03-27:
Still happening.
Not only is the checkbox black, but the text is half hidden and the checkbox doesn't have a legend.
Linux Mint 21.3 Cinnamon

YoSTEALTH, 2024-08-21:
I was having the same issue, I use manjaro (linux) so solved it by:
`System Settings` -> `Manjaro Settings Manager` -> `Locale Settings` -> `Add` -> select `English` + `United Kingdom` -> `Add`

That should do it, good luck.

andrew-4D, 2024-10-22:
FYI, black checkbox = "unselected"  white checkbox = "selected"

mike-lloyd03, 2024-10-25:
Depending on your theme, the checkbox might be completely invisible. I just clicked around to find it.
[image]
[image]

MackBambu, 2024-11-18:
fixed

closed 2024-11-18 by MackBambu