bambulab/BambuStudio #2712: New Studio 1.7.7.88 doesn't purge enough on startup

kazolar, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Windows 10 

### Printer

Bambu Lab X1C 2 AMS units via a hub

### How to reproduce

Have PLA loaded in slot 1 of AMS 1
First print was badly underextruded. Nothing special was done -- see attached project file.
Then triggered load filament from printer screen, purged some filament, printed again. Was fine
Then printed another version straight from studio, again badly underextruded.
Again did the load filament and purged some, and it printed fine. NOT ENOUGH purged on startup

### Actual results

Bad underextrusion

### Expected results

Should work as previous versions. Please stop rolling out badly tested updates. 

### Project file & Debug log uploads

[jig2040V2.zip](https://github.com/bambulab/BambuStudio/files/12887279/jig2040V2.zip)
[studiolog.zip](https://github.com/bambulab/BambuStudio/files/12887283/studiolog.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kazolar, 2023-10-12:
Orca Slicer works fine. Not sure what you guys did here, but last 2 versions have been really bad

QingZhangBambu, 2023-10-13:
[image]
[image]

the machine start gcode is too old, and may cause nozzle clog.  could you use the new one and try?

kazolar, 2023-10-13:
I reverted printer firmware. The firmware is really glitchy, can't cancel prints, can't communicate with slicer. Re-ran the print with PETg and it's fine. Trying different PLA. Might be that PLA spool.

kazolar, 2023-10-13:
So I bumped the temp for the same black pla to 230 and it printed fine. Had been set to 220 for many other prints. Could be part is small or some other reason. Will stick to the old firmware and Orca Slicer until bambu firmware and studio get through beta phase of new printers being added

Xiuqi-Huang-CN, 2023-10-16:
Maybe the extruder wheels and/or nozzle are blocked by mistake, or the temperature control or the thermal conductivity of the nozzle work badly, and lead to lower actual temperatures than preset ones, causing the extrusion resistance force to be larger, and resulting in underextrusion, so a higher nozzle temperature 230 ℃ works well than the default 220 ℃. You can first try to clean the extruder and the nozzle, or replacethe nozzle with a new one, and check if the problem can be solved.
You can refer to this wiki:
[(https://wiki.bambulab.com/en/knowledge-sharing/common-print-quality-problem)]
[image]
Wish you good luck!

Xiuqi-Huang-CN, 2023-10-16:
@kazolar