bambulab/BambuStudio #2713: Generic PLA High Speed can no longer be selected in AMS

LTB3D, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.0

### Printer

X1 Carbon

### How to reproduce

1. Slice Place
2. Print Plate

### Actual results

Generic PLA High Speed now shows up as PLA+ and is no longer selectable in the AMS (only with the Bambu_Studio_mac-v01.07.07.88, it was fine with Bambu_Studio_mac-v01.07.06.92)

[image]

### Expected results

[image]

### Project file & Debug log uploads

[Bug.zip](https://github.com/bambulab/BambuStudio/files/12887357/Bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-10-13:
Generic PLA High Speed is filament profile for PLA. do you mean the filament detect is wrong? PLA be identified as PLA+?

LTB3D, 2023-10-13:
No, I would not say that it is "detecting" anything. **Generic PLA High Speed** is the profile that I had/have selected for slots A1 & A4. I'm using HS filament from another brand in those slots. 
The upgrade to 1.7.7.88 changed the profile somehow & now it is calling slots A1 & A4 PLA+.

That on i's own would be fine, but it's not allowing me to select slots A1 and /or A4 at all & automatically mapping it to slot A3. Only slots A3 & A2 come up as available. The files/designs were prepared as Filament # 1 (aka A1)
[image]

[image]

But when I hit "print plate" it changes it remaps it to Filament # 3 (aka A3)
[image]

And does not allow me to select slots A1 or A4 at all
[image]

I've already hit "sync" and "re-sync", but it didn't help. It's like the Profile for HS PLA was updated in one part of the software (Device tab) but not in the other sections of the software (Prepare & Preview tabs) and now it is confused. 
I'm thinking the "Device" is now calling it PLA+, but "Prepare" is still calling it PLA.

Does that make sense?

StoneLiBambulab, 2023-10-13:
@sodachen22 

printer AMS push a PLA+ type when select the Generic PLA High Speed, please have a check.

Thanks!

Stone Li（旺仔）

sodachen22, 2023-10-13:
@LTB3D It looks like there is problem in filament configuration after Studio upgrading from v01.07.06.92 to 01.07.07.88. Would you please share the configuration files in json format in this directory?
[image]

LTB3D, 2023-10-13:
[filament.zip](https://github.com/bambulab/BambuStudio/files/12892604/filament.zip)

StoneLiBambulab, 2023-11-06:
There is a bug when getting the filament type, will be fixed in 1.8.xx

Thanks!
Stone Li

StoneLiBambu, 2024-01-29:
fixed

closed 2024-01-29 by StoneLiBambu