bambulab/BambuStudio #2714: "Plate Type" is greyed out for 3rd Party printer and cannot be changed.

jonathangregsonrogerscom, 2023-10-12:
### Bambu Studio Version

1.7.7.88

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Following Bambu Lab wiki doc I created 3rd party printer. Seems to be working fine except the "Plate Type".  My printer only has a cool plate.  I've reviewed all json files and cannot find a reference setting for plate_type.  As it is, the printer will print but throws a warning mssg indicating that since the printer doesnt have a heated bed all further cmmds re heated bed will be ignored.

### Printer

3rd party printer - XYZ Printer daVinci Mini w+ 

### How to reproduce

1. follow wiki instructions on adding 3rd party printer.
2. launch Bambu Suite and add printer.
3. Try to change Plate Type to "Cool Plate"

### Actual results

Add 3rd party printer using wiki instructions.

### Expected results

When Bambu Suite loads 3rd party printer should have Plate Type not greyed out.

### Project file & Debug log uploads

[daVinci Mini w+.zip](https://github.com/bambulab/BambuStudio/files/12887284/daVinci.Mini.w%2B.zip)
[debug_Thu_Oct_12_16_43_31_10956.log.zip](https://github.com/bambulab/BambuStudio/files/12887366/debug_Thu_Oct_12_16_43_31_10956.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-10-13:
[image]
if the cool_plate_temp be set to 0, means the plate not be used, otherwise, this plate can be used.

jonathangregsonrogerscom, 2023-10-13:
Thx for the quick reply!! I'll give it a try this morning.

Thx
Jonathan

Sent from Rogers Yahoo Mail on Android

jonathangregsonrogerscom, 2023-10-13:
I checked and there is no Cool Plate temp avail to change (see attached).

[image]

QingZhangBambu, 2023-10-16:
i checked the code logic, 3d party printers do not distinguish plate type. just modify the plate temp, it can be used as a cool plate.

SaltWei, 2024-01-17:
For 3rd party printer, directly set the bed temperature. Not a bug

closed 2024-01-17 by SaltWei