bambulab/BambuStudio #2717: Allow printer to remember previous filament and not purge if new print is using old filament.

davidja92, 2023-10-13:
**Is your feature request related to a problem? Please describe.**
time and filament is wasted when every print retract filament to ams even if next print will use the same filament type and color 

**Describe the solution you'd like**
Allow printer to remember previous filament and not purge if new print is using old filament. this would eliminate the cutting and retracting of filament at the end of every print. thus saving time and filament if printing rapidly with same filament.

**Describe alternatives you've considered**
i have tried to create my own start and end gcode but the printer does not recognize the commands

**Additional context**
attached i modified the start g code to have the printer call for the previous_extruder (the last filament used) and cut retract and purge if filament is not equal to new print {if filament_type[initial_no_support_extruder]!="previous_extruder"}

or bypass the purge {elsif (filament_type[initial_no_support_extruder]=="previous_extruder") } if the filament is the same.
[start g code with retract if filament different.txt](https://github.com/bambulab/BambuStudio/files/12888615/start.g.code.with.retract.if.filament.different.txt)

pichen2021, 2023-10-13:
Hi @davidja92 
This feature may need to be implemented in firmware.
"previous_extruder" can be obtained from the machine when the studio is slicing and replaced when generating gcode, but printing may also be initiated directly from the SD card, app, etc. The printing file cannot be re-sliced and the generated .gcode will be used directly, but "previous_extruder" may have changed.
We will evaluate this requirement in future firmware.
Thank you.

Geekay1972, 2023-11-20:
I would go even further ... why is there an automatic unload in the AMS after a print is finished? Interesting enough it only happens when printing from the AMS. If you print from external spool the filament keeps loaded.

I was also annoyed as I'm printing most of the time from one spools so it's not only a waste of filament but also of time.

Wookbert, 2023-12-12:
Remembering the previous filament type & color beyond filament eject and insertion _– thus skipping the Nozzle purge –_ is a feature my CEL Robox FDM printer (with its CEL Automaker Slicer) has/had **since 2015.**

juamiso, 2024-05-17:
This is BY FAR the most annoying thing that my Bambu lab a1 mini is missing. It is really a pain in the as to iterate on designs of small pieces that you reprint every 10 minutes to see how the stupid printer is retracting, cutting, purging old filament, reoading, ... just to make a new print with exactly the same filament. Waste of time and money. Please implement this, is such an easy "win" !