bambulab/BambuStudio #2720: Bambu slicer load filament issue

Crobilt, 2023-10-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

When I push the load button in Bambu slicer it doesn't do anything. Unload button works fine

### Actual results

Nothing

### Expected results

Filament should load

### Project file & Debug log uploads

no file available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lavinir, 2024-03-16:
Was anything ever done with this? I seem to be having the same issue

Crobilt, 2024-03-18:
No. I still can not unload filament. 🦆 Cro