bambulab/BambuStudio #2721: supports problem

justindzs, 2023-10-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Edition	Windows 11 Home Version	22H2 Installed on	‎10/‎16/‎2022 OS build

### Additional system information

Device name	DESKTOP-U09PDJC
Processor	Intel(R) Core(TM) i9-10900 CPU @ 2.80GHz   2.81 GHz
Installed RAM	32.0 GB (31.8 GB usable)
Device ID	B387C7C8-0550-47BF-8E0F-866873FA1463
Product ID	00325-82111-77059-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen support

### Printer

p1s

### How to reproduce

using add object to raise my project off the build plate.
then sliced with both normal supports and tree supports

### Actual results

the normal supports go all the way to the build plate as they should the tree supports act like the model is still on the plate so don't go all the way down to the actual build plate they just hang in the air.

### Expected results

base of the tree supports should reach the build plate  simply adding a button to turn of the auto snap to plate would probably be a good idea like most slicers do.

### Project file & Debug log uploads

na

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2023-10-24:
3mf file please

justindzs, 2023-10-26:
The problem was that when suspended normal supports would reach the build
plate but tree supports thought it was still on the build plate and would
be suspended in the air. just went to get you a file and it seams t be
working correctly now. Thank you for getting back to me.

closed 2023-11-24 by lanewei120