bambulab/BambuStudio #2723: Bambu Studio 1.7.7.89 closes when print is sent to printer.

hbar98, 2023-10-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windoes 11

### Additional system information

OS Name	Microsoft Windows 11 Pro
Version	10.0.22621 Build 22621
System Manufacturer	LENOVO
System Type	x64-based PC
System SKU	LENOVO_MT_20L8_BU_Think_FM_ThinkPad T480s
Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz, 1800 Mhz, 4 Core(s), 8 Logical Processor(s)
Installed Physical Memory (RAM)	24.0 GB
Adapter Description	Intel(R) UHD Graphics 620
Adapter RAM	1.00 GB (1,073,741,824 bytes)
Driver Version	30.0.101.1122
Resolution	1920 x 1080 x 60 hertz

### Printer

X1C

### How to reproduce

1. Open Bambu Studio
2. Open a model, adjust settings, and slice
3. In Bambu Studio, send sliced file to printer to print
4. Bambu Studio closes with no way to interrupt and no warnings it is closing

### Actual results

Bambu Studio closes with no dialogue boxes and no warning it is going to close.

### Expected results

The program should remain open so I can click on the Devices tab to view progress or work on another model. 

### Project file & Debug log uploads

[debug_Fri_Oct_13_14_46_22_4160.log.zip](https://github.com/bambulab/BambuStudio/files/12898356/debug_Fri_Oct_13_14_46_22_4160.log.zip)

[bambu studio crashes_pixel-8-proV2.1.zip](https://github.com/bambulab/BambuStudio/files/12898432/bambu.studio.crashes_pixel-8-proV2.1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-10-16:
@hbar98
Is it a 100% crash in this situation?

hbar98, 2023-10-16:
With that model, yes. But I've done other models since then and it hasn't
crashed or closed. I haven't tried to print that model since then.

jonathangregsonrogerscom, 2023-10-16:
ver 1.7.7.88 does the same on occasion.  Similar OS & HW.

bambu123, 2023-10-17:
> ver 1.7.7.88 does the same on occasion. Similar OS & HW.

Please help provide logs. Also, whether it is a 100% failure.

bambu123, 2023-10-17:
> > ver 1.7.7.88 does the same on occasion. Similar OS & HW.
> 
> Please help provide logs.

bambu123, 2023-10-17:
> With that model, yes. But I've done other models since then and it hasn't crashed or closed. I haven't tried to print that model since then.

To analyze DMP files, we need you to provide the following DLLs on your system: 

C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.22621.2215_none_270d72677387dfc3\comctl32.dll
C:\Windows\System32\ntdll.dll
C:\Windows\System32\kernel32.dll
C:\Windows\System32\KERNELBASE.dll
C:\Windows\System32\user32.dll
C:\Windows\System32\msvcp140.dll
C:\Windows\System32\ucrtbase.dll
C:\Windows\System32\win32u.dll

bambu123, 2023-12-25:
We have released version v1.8, please try using the new version to see if the issue exists.