bambulab/BambuStudio #2725: Nag warning regarding opening door on printer every print initiated from the printer's screen.

MrTantrum, 2023-10-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 Series

### How to reproduce

- In Bambu Studio slice a file
- Select SEND from top right drop down
- Click SEND button to send job to printer
- From printer touch screen select job and initiate print

### Actual results

At this point the nagging caution message about opening the print door appears.

### Expected results

There should not be an unnecessary message posted on the printer's display EVERY PRINT that the user has to push to clear.

### Project file & Debug log uploads

There is not one.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-10-16:
Do you get the warning on the studio or on the printer? in version .89 , it supported putting it on the daily tips.

sodachen22, 2023-10-31:
According to the design, this prompt will appear every time you print to prevent users from forgetting and causing clogs

MrTantrum, 2023-10-31:
> Links:

First, the prompt is an incredible nag.  I print multiple items every 
day and all this does is get in the way, results in an extra click, and 
if the user has to be reminded every single print then maybe they 
shouldn’t be using this printer.

Secondly, the information is just wrong.  I’ve printed dozens of models 
(mostly multicolor) using numerous rolls of PLA, nonstop for months all 
with the door closed and even with the fan speed turned down to lower 
the noise level, and this just isn’t an issue.

It is extremely frustrating to have to click on this pass through screen 
every time.  You have recently added this “feature” that nobody asked 
for, and you no it is such a nag that you have already removed it from 
appearing in the slicer every print.  I’m just asking you do the same 
when direct printing.  At least give the operator the ability to either 
enable or disable this warning in settings if you insist on keeping it.

Quite honestly, I love this printer, but this one purposefully 
introduced, unnecessary nag is enough to make me complain about it every 
chance I get.

sodachen22, 2023-11-02:
Understand, there may need to be an option to disable the notification here, we will consider the solution.

MrTantrum, 2023-11-02:
Thank you, I would appreciate anything you are willing to do.

GuzziRaz, 2023-11-06:
Also consider #2751 - no need to warn if door is already open. But even with that in place we still want an option to disable the warning completely in case, for example, the lid is permanently off.

Also, I agree the warning is questionable. I always run with lid and door closed, to reduce noise. Instead I monitor the chamber temperature. If it would get near 40°C and full speed fan wouldn't help, sure I would open the lid. It's just that it hasn't ever happened to me!

IMHO the printer should start out with chamber fan very low (25% or less, again for low noise) and automagically adapt fan speed during the entire print. **If** chamber fan is already at 100% and temperature is still getting dangerously close to softening temperature - only THEN I would like to get a warning notification in Studio and Handy.