bambulab/BambuStudio #2727: Cannot print after update to Bambu Studio 1.7.7.89

intrinsic420, 2023-10-14:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.4.1

### Printer

Bambu Lab X1C

### How to reproduce

This happened directly after updating Bambu Studio.I loaded a STEP file to print using Generic PETG settings on a textured PEI sheet. Supports on base plate only. I send it to the X1C using the AMS slot 4 where PETG is loaded. When the slicer uploads it to my printer, it says "The selected printer is incompatible with the chosen printer presets." I have changed absolutely nothing and literally printed something within the past couple of days with the exact same settings, just the older version of the slicer.

### Actual results

[image]
Here's a screenshot of the dialogue box.

### Expected results

It should print like it always did before.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12902357/log.zip)
[bike grip adapter.3mf.zip](https://github.com/bambulab/BambuStudio/files/12902358/bike.grip.adapter.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

intrinsic420, 2023-10-15:
Further testing: attempted print of Bambu Lab's wireless mouse top shell using the presets and raising bed and nozzle temp slightly for silk PLA. Attempted to print on both Cool Plate and Textured PEI sheet. Both say that my printer is not compatible. Loaded file to MicroSD card and printed from there. I am now getting layer shifts in the print. Very inconsistent. The last attempt tried printing a layer about 10cm away from where the print was at about the 10th layer.

I also tried all of this from a PC running the latest Windows updates and the latest slicer version.

Also, since the slicer update my print head BANGS against the stops during the startup procedure. Whenever it homes the head, or goes to the purge slot, it bangs around hard enough now to where the print head cover is coming off. Interestingly, the cover warning hasn't come up at all, even when I took it completely off to reseat the magnets. I did that while the head was moving accidentally. No warning. I want to revert my slicer version but I don't know how :(

QingZhangBambu, 2023-10-16:
[image]
[image]

The pinter in 3mf be chosen is A1 mini, it is not compatible  printer of X1C. Because the popup is in the left, it may damage X1C's extrude.

QingZhangBambu, 2023-10-16:
and it should be pohibited to use A1 mini profile in X1C machine. @sodachen22

intrinsic420, 2023-10-18:
Thanks Qing! I feel stupid not checking that, but to be fair I only had to set that parameter once on initial install.

closed 2023-10-18 by intrinsic420

sodachen22, 2023-10-31:
Printer will support the feature that prevent the printing job if the sliced 3MF is not compatible with the printer.