bambulab/BambuStudio #2728: AMS Cabin Humidity indicator for colorblind

Th3M1k3y, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
I am having problems seeing the color on the indicator for AMS Cabin Humidity

**Describe the solution you'd like**
Being able to hover the mouse over the indicator and then get a tooltip with Red/Yellow/Green

**Describe alternatives you've considered**
Adding H for high, M for medium, L for low to the icon.

**Additional context**
[image]

EStreet706, 2023-10-16:
I second this.  Also, selecting videos/models to delete is impossible for colorblind people to see correctly.   That faint garbage is just that.  Don't use it!

walterwongbbl, 2023-10-17:
Hi. @BlackNet @Th3M1k3y 
We have already provided feedback to the designers regarding this issue.

BlackNet, 2023-10-18:
Well. New version does have changes so kudos for that. Light mode the problem is still there as just a background change is not enough. The icon itself is faint while everything else around it is higher contrast.

walterwongbbl, 2024-03-11:
We have optimized this issue and it will be closed

closed 2024-03-11 by walterwongbbl