bambulab/BambuStudio #2729: Color-1 Infill in Color-2 area of print.

kzemlin, 2023-10-14:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

OS Name	Microsoft Windows 11 Pro
Version	10.0.22621 Build 22621
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	ZEMLINP16
System Manufacturer	LENOVO
System Model	21D6004XUS
System Type	x64-based PC
System SKU	LENOVO_MT_21D6_BU_Think_FM_ThinkPad P16 Gen 1
Processor	12th Gen Intel(R) Core(TM) i9-12900HX, 2300 Mhz, 16 Core(s), 16 Logical Processor(s)
BIOS Version/Date	LENOVO N3FET37W (1.22 ), 7/28/2023
SMBIOS Version	3.4
Embedded Controller Version	1.10
BIOS Mode	UEFI
BaseBoard Manufacturer	LENOVO
BaseBoard Product	21D6004XUS
BaseBoard Version	SDK0T76528 WIN
Platform Role	Mobile
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\Windows
System Directory	C:\Windows\system32
Boot Device	\Device\HarddiskVolume3
Locale	United States
Hardware Abstraction Layer	Version = "10.0.22621.1413"
User Name	ZemlinP16\karl
Time Zone	US Eastern Daylight Time
Installed Physical Memory (RAM)	64.0 GB

### Printer

Bambu Labs X1 Carbon

### How to reproduce

Open project.
Slice plate.

### Actual results

Build should require 5 color changes.  When sliced there are 171.  Slicer is adding blue infill/solid layers in yellow sections of the build.  Currently the project assigns colors by height ranges.  Tried painting surface fills first, but results were the same.
[image]

### Expected results

Build should only require 5 color changes.

### Project file & Debug log uploads

[Elder_Wand_tip_Bambu v2.zip](https://github.com/bambulab/BambuStudio/files/12906865/Elder_Wand_tip_Bambu.v2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kzemlin, 2023-10-15:
[debug_Sun_Oct_15_10_55_45_41824.log.zip](https://github.com/bambulab/BambuStudio/files/12909864/debug_Sun_Oct_15_10_55_45_41824.log.zip)

zhimin-zeng-bambulab, 2023-10-18:
Thank you for reporting this issue, we will fix it as soon as possible

zhimin-zeng-bambulab, 2024-01-23:
[image]
@kzemlin  There are hollow holes inside this model, for example, after I cut it at a height of 54.2, I can see a small hole on the cutting surface. The inner wall of the hole is not painted, so the default color is blue, while the outer wall It is yellow, so there will be a color change. 
Currently we will only delete holes with an area smaller than 0.01mm2.

zhimin-zeng-bambulab, 2024-02-04:
@kzemlin  Does the above explanation give you a clear understanding of the reasons?

kzemlin, 2024-02-04:
It's not clear to me how internal holes would impact painting by height ranges, but I'm willing to accept that there are geometry flaws causing the issue.  Thank you.

zhimin-zeng-bambulab, 2024-02-19:
@kzemlin  Yes, in theory, the internal holes should also be painted, but it may be due to flaws in the mesh data of the model, which causes this phenomenon.
You can right-click the object, select "Fix model", then repaint it, and then slice it.  This can avoid the problem.
[image]

zhimin-zeng-bambulab, 2024-03-08:
@kzemlin Can this problem be solved by fixing the model?

zhimin-zeng-bambulab, 2024-04-29:
Since the provided model has unpopular edges, the model needs to be repaired before slicing. It not a bug. Close it.

closed 2024-04-29 by zhimin-zeng-bambulab