bambulab/BambuStudio #2731: Linux Dark Mode Process pane object names illegible

keithel, 2023-10-14:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 23.04 Lunar Lobster

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i7-13700H
Memory: 64GB
Graphics Card: RTX A1000 6GB Laptop GPU

### Printer

Bambu Lab P1P

### How to reproduce

1. Open Gnome settings
2. Click Appearance tab
3. Choose `Dark` style
4. Color is set to the default Ubuntu installs with - the leftmost choice of orange.
5. Open Bambu Studio
6. Load any project with objects.
7. In the left bottom `Process` pane, in the `Global/Objects` mode toggle, switch to `Objects` properties mode.
8. The bug should present itself.

### Actual results

After following all steps, the names of the plates and names of the objects are all in a font color that is extremely difficult to read on the dark grey background. The font color is just slightly different than the dark mode background color. The checkboxes for the object and assembly visibility in the second column are a different shade of grey, a little easier to see than the object, assembly, and plate names, but is also far too dark.

### Expected results

After following all reduction steps, the plate, assembly, and object names in the `Process` pane should be rendered in a legible font color. I think a lighter shade of grey would probably be best, but that is up to you. All that really needs to happen is that it has enough contrast on the background to be legible.

[image]

### Project file & Debug log uploads

Bambu Studio Project file I was using:
[Tesla J1772 Holder short.zip](https://github.com/bambulab/BambuStudio/files/12907873/Tesla.J1772.Holder.short.zip)

Log file in this case I think is unnecessary. If you need it, let me know and I can attach one.

### Checklist of files to include

- [ ] Log file
- [X] Project file

keithel, 2023-10-14:
Workaround for this is to just use light mode. Unfortunately, the Linux release of Bambu Studio does not contain a dark/light mode toggle in the app so that Bambu Studio can be in light mode while the rest of your desktop is in dark mode. That issue was reported as #1357.

Cass67, 2023-10-24:
Seems ok on Pop_OS 22.04
[image]

keithel, 2023-10-24:
I can still reproduce this with v1.7.7.89 Ubuntu AppImage release.

flurl, 2023-12-08:
I have the same problem since I started using Bambu Studio. Unreadable text in dark mode for some parts of the slicer. It is somehow usable, but very annoying. Same problem with either Ubuntu or Debian with the Ubuntu app image. 
As a workaround startin it this way works for me:
`GTK_THEME=Adwaita:light ./BambuStudio_linux_ubuntu_v01.08.01.57.AppImage`

keithel, 2023-12-09:
I haven't figured out how to build Bambu Studio from sources yet, as the prerequisites listed in the linux compile wiki page are out of date, and I'm stuck on CMake not finding OpenVDB (even though I have it installed) - so I can't make the change needed to the sources myself yet.

In the interim, one can unpack the AppImage, make the needed change, and repack the AppImage to get one that will always start the app with the Adwaita:light theme:

1. cd to a temp directory
2. Extract your Bambu_Studio[...].AppImage using: `.../Bambu_Studio[...].AppImage --appimage-extract`<br>This will extract the AppImage into a directory called `squashfs-root`.
3. Edit the file `squashfs-root/AppRun`
4. This file is a shell script that ultimately launches Bambu Studio. In here, add:<br>`export GTK_THEME=Adwaita:light` before the `exec` command.
5. Save and exit the editor.
6. Download the `appimagetool` AppImage found here under `Assets`: https://github.com/AppImage/appimagetool/releases and make it executable.
7. Run `<path to appimagetool>/appimagetool-x86_64.AppImage squashfs-root`<br>This will repackage the AppImage with the name `BambuStudio-x86_64.AppImage`.
8. [Optional] Rename this back to the original name of the AppImage, perhaps adding something to the name to indicate that you've modified it.
9. Remove/move aside the original Bambu_Studio AppImage, and put this new one in its place.<br>I use AppImageLauncher, and have the directory $HOME/AppImages to hold all of my AppImages. If you use AppImageLauncher, after you place the newly repackaged Bambu_Studio AppImage in the directory, give it 15-30 seconds or so for AppImageLauncher to recognize it and provide it to the desktop manager.
10. You're done - just launch BambuStudio as you have been. It will now always come up in the light theme (adwaita-light).

maruel, 2024-02-21:
I came here to say the same thing but with the light theme (!); I'm on Ubuntu 22.04 on Bambu Studio v1.8.4.51. Here's how it shows up with a light theme:
```
export GTK_THEME=Adwaita:light
./Bambu_Studio.AppImage
```

[image]

Here's how it shows up with with dark theme:
```
export GTK_THEME=Adwaita:dark
./Bambu_Studio.AppImage
```
[image]

I tried other themes (HighContrast, Yaru, Raleigh) from glancing at /usr/share/themes but it just seems to revert to Adwaita:light which is the default on my computer.

MilesRdz, 2024-05-26:
[log.zip](https://github.com/bambulab/BambuStudio/files/15448692/log.zip)
I'm running into this as well.
I've attached the log and config file.

t-fi, 2024-06-30:
I observe the same as @maruel . Switching to light mode does fix some dialogues. But specifically the print settings transfer dialogue is unreadable in both dark and light mode.

token47, 2024-09-06:
You can edit your BambuStudio.desktop file and add an override there instead:

```
[Desktop Entry]
Name=BambuStudio
Exec=env GTK_THEME=Adwaita:light SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt /opt/bin/Bambu_Studio_ubuntu-v01.09.05.51-20.04.AppImage %F
Icon=BambuStudio
Terminal=false
Type=Application
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;
Categories=Utility;Graphics;3DGraphics;
Keywords=3D;Printing;Slicer;

```

psxpaul, 2024-11-03:
I'm seeing this issue on Arch. Using Flatseal to set the `GTK_THEME=Adwaita:light` env variable seems to also work as a workaround.

token47, 2024-11-04:
The problem is that in light mode it is a little more readable but still not good. Example below:
[image]

mike-lloyd03, 2025-02-07:
This happens when importing STEP files as well.

[image]

maruel, 2025-03-25:
Just a note that the problem is fixed for me with 1.10.2 on Ubuntu 22.04 with light theme. Thanks!

jimisola, 2025-03-25:
> Just a note that the problem is fixed for me with 1.10.2 on Ubuntu 22.04 with light theme. Thanks!

The issue is with dark mode according the title. Was there ever an issue with a light theme or only with dark themes?

MackBambu, 2025-03-26:
Fixed in v2.0

closed 2025-03-26 by MackBambu