bambulab/BambuStudio #2732: Bambu Studio 1.7.7.89 Will not open

Fowlerman24, 2023-10-14:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Edition	Windows 11 Home Version	22H2 Installed on	‎10/‎23/‎2022 OS build	22621.2428 Serial number	4WNM9FGR600190 Experience	Windows Feature Experience Pack 1000.22674.1000.0

### Additional system information

Processor	11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   2.80 GHz
Installed RAM	16.0 GB (15.7 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Click on desktop icon. Does nothing. Tried re-installing, did not help. 

### Actual results

Nothing

### Expected results

App should open

### Project file & Debug log uploads

?

### Checklist of files to include

- [X] Log file
- [ ] Project file

Fowlerman24, 2023-10-16:
> @Fowlerman24 <https://github.com/Fowlerman24>
>
>    1. Is previous version workable on your windows?
>    2. after you click the desktop icon, is the splash screen popup?
>    3. could you upload the files under C:\Users\user.name\AppData\Roaming\BambuStudio\log
>    for us to have a check?

I never had the previous version due to I just got the printer. I tried
downgrading versions to get to work and still have no luck with it.

lanewei120, 2023-10-16:
> * after you click the desktop icon, is the splash screen popup? could you record a short video for this process?
> * could you upload the files under C:\Users\user.name\AppData\Roaming\BambuStudio\log for us to have a check?

after you click the desktop icon, is the splash screen popup? could you record a short video for this process?

could you upload the files under C:\Users\user.name\AppData\Roaming\BambuStudio\log for us to have a check?

Fowlerman24, 2023-10-16:
I'm currently out of town. I will record when I get home. But until then,
what is happening is when I install the software, its showing it's
installed on my computer. But when I click on the icon, it spins for 1
second and never opens. No, pop-up screens, nothing. It's almost like the
application is dead Or unrecognizable. I originally thought maybe windows
defender was blocking it somehow. No luck there

Fowlerman24, 2023-10-19:
Sorry, I might be a little computer illiterate. But I'm not sure how to upload the file. I have a short video

lanewei120, 2023-10-20:
> Sorry, I might be a little computer illiterate. But I'm not sure how to upload the file. I have a short video

@Fowlerman24 
how about using https://www.dropbox.com?

Fowlerman24, 2023-10-20:
Hopefully this works for you.

https://www.dropbox.com/scl/fi/3255ss5fs3rois5r9enhf/20231019_182817.mp4?rlkey=aqtzxy4r1efvvrmivz9idem3d&dl=0

lanewei120, 2023-10-20:
@Fowlerman24 

thanks

Is there any window popup in the other monitor?
and could you help to connect only one monitor and try again, we need to check whether it is related with dual monitor or not

also there is a directory C:\Users\**your_user_name**\AppData\Roaming\BambuStudio\log
please help to packed them to a zip and upload here

Fowlerman24, 2023-10-20:
No nothing is popping up on the other monitor. I have already tried this on
single monitor and just on laptop. Still doing the same thing. I apologize
i'm not familiar with the directory portion. Is there a way I can provide a
phone number and you can walk me through it?

lanewei120, 2023-10-20:
> No nothing is popping up on the other monitor. I have already tried this on single monitor and just on laptop. Still doing the same thing. I apologize i'm not familiar with the directory portion. Is there a way I can provide a phone number and you can walk me through it?

@Fowlerman24 
how about using discord?
my discord account is lhwei121@163.com

Fowlerman24, 2023-10-20:
So good news it looks like I might have found the issue. When going through
the program folder I came across a plugin folder. Clicked on
vcredist2019_x64 and hit repair. After the computer restarted it opened.

lanewei120, 2023-10-20:
> So good news it looks like I might have found the issue. When going through the program folder I came across a plugin folder. Clicked on vcredist2019_x64 and hit repair. After the computer restarted it opened.
> […](#)

Glad to hear that

lanewei120, 2023-10-24:
close this issue as solved

closed 2023-10-24 by lanewei120