bambulab/BambuStudio #2733: [BUG] Useless Block Support Enforcers still not working on latest BambuStudio

bolsoncerrado, 2023-10-15:
### Bambu Studio Version

1.7.7.8.9

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura

### Additional system information

M1Ultra

### Printer

X1C, P1P

### How to reproduce

1. Create a job
2. Enforce support blockers like in the pics
3. Preview.

### Actual results

4. Supports are not blocked

### Expected results

Supports....blocked?

### Project file & Debug log uploads

[image]
[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bolsoncerrado, 2023-10-15:
Might be related to July's issue: https://github.com/bambulab/BambuStudio/issues/2019

tangketan, 2023-10-24:
3mf file please.

bolsoncerrado, 2023-10-24:
@tangketan sure, is there a way to share the file "privately"?

tangketan, 2023-10-24:
@bolsoncerrado You may email me: arthur.tang at bambulab.com

tangketan, 2024-03-29:
@bolsoncerrado Sorry for the late reply. Your 3mf has both enforcers (painted supports) and blockers, and enforcers have higher priority.

tangketan, 2024-03-29:
I'm closing this issue since it's not a bug. But feel free to re-open it if you'd like more discussion.

closed 2024-03-29 by tangketan