bambulab/BambuStudio #2734: [FEATURE REQUEST] BedCheck AI

bolsoncerrado, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
When owning multiple printers and in "high occupancy" season sometimes you MAY forget a print (or a tool?) in the bed. If you launch a print, terrible things (and sounds!) happen!

**Describe the solution you'd like**
Implement an AI algorithm so it takes a snapshot of the current bed and compares with a "clean" bed and reports "are you sure the bed is clean?" pausing the ongoing print start....

**Describe alternatives you've considered**
https://plugins.octoprint.org/plugins/bedcheckai/

In fact, the screenshots on that plugin show a BambuLab printer so... you're being late to the party! :P

sodachen22, 2023-10-31:
Thanks for the feedback. We will investigate and confirm the feasibility.

plebbyd, 2023-11-28:
Original developer of the BedCheck AI ML model here. If BBL is interested in this feature please contact me at: lebiedzinskip@printpal.io

bolsoncerrado, 2025-06-02:
Why is this not supported yet!??!