bambulab/BambuStudio #2735: BambuStudio for Mac camera stream playback issue

Magalex2x14, 2023-10-15:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.0

### Printer

Bambu Lab X1CC

### How to reproduce

Just trying to watch a live stream from the printer's camera.

### Actual results

[This video clearly demonstrates the problem.](https://youtu.be/VD8vKUHor2M) Above is playback in BambuStudio for Mac 1.7.7.89 (Sonoma 14.0), below is a virtual machine window on the same computer with BambuStudio 1.7.7.89 for Windows (Windows 10 Parallels virtual machine). Just like on Windows, Bambu Handy for iOS also plays video stream normally. I also tried to open the stream via the link rtsps://bblp:<lanpass>@<lan ip>:322/streaming/live/1 using the Movist Pro player - the stream plays normally (however it stops after about 10 seconds, but it doesn’t matter ).

### Expected results

Smooth quality playback.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12910527/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Magalex2x14, 2023-10-15:
I note that there are users on the forum [with the same problem](https://forum.bambulab.com/t/glitchy-video-stream-from-x1c-into-bambu-studio).

arekm, 2023-11-02:
For me stream shows old and new frames repeatably all the time in BambuStudio on MacOS with P1S:

https://www.youtube.com/watch?v=89pXEO3CJus
(note the clock and seconds)

bambu123, 2023-11-03:
@arekm @Magalex2x14  Is the old version of BambuStudio normal?

arekm, 2023-11-03:
> Is the old version of BambuStudio normal?

"It was happening on older firmware and older Bambu Studio, too." in video description on youtube.

But I can test again some specific old versions, which one for example?

bambu123, 2023-11-03:
> > Is the old version of BambuStudio normal?
> 
> "It was happening on older firmware and older Bambu Studio, too." in video description on youtube.
> 
> But I can test again some specific old versions, which one for example?

Please test the 1.7.3 version.
[v1.7.3 (v1.7 Hotfix 2)](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.03.50)

arekm, 2023-11-03:
1.7.3.50 behaves much better. I think it still has some old-new frame rare jumps but that's like 1s jump (in 0m7s of video below). And no "green screen" effect.

https://www.youtube.com/watch?v=Q-d5UZv0P1c
(in 4k but youtube still processes it, so 240p only for now; can take up to 2h)

Bambu Studio 1.7.7.89 on top
Bambu Studio 1.7.3.50 on bottom

At 4m0s (https://youtu.be/Q-d5UZv0P1c?t=238) I'm stopping stream and starting again. 1.7.7.89 has trouble to restart it. 

At 5m7s (https://youtu.be/Q-d5UZv0P1c?t=304) I'm restarting 1.7.7.89 stream again and this time it restarts fast.

bambu123, 2023-11-06:
> 1.7.3.50 behaves much better. I think it still has some old-new frame rare jumps but that's like 1s jump (in 0m7s of video below). And no "green screen" effect.
> 
> https://www.youtube.com/watch?v=Q-d5UZv0P1c
> 
> Bambu Studio 1.7.7.89 on top Bambu Studio 1.7.3.50 on bottom
> 
> At 4m0s (https://youtu.be/Q-d5UZv0P1c?t=238) I'm stopping stream and starting again. 1.7.7.89 has trouble to restart it.
> 
> At 5m7s (https://youtu.be/Q-d5UZv0P1c?t=304) I'm restarting 1.7.7.89 stream again and this time it restarts fast.

Thanks for the feedback. For P1P and P1S, there are indeed some changes in these two versions. Different data channels are used, and there may be problems with the data of the printer LAN channel. However, for X1C, there are some differences in the phenomenon, and there are no version changes. The reasons need further investigation.

arekm, 2023-11-16:
Recently it was discovered how to access camera stream directly from P1P/P1S printer, so I tested that:

https://github.com/bambulab/BambuStudio/issues/1536#issuecomment-1812001751

and stream from P1S starts immediately (within 2.5s), is stable, no artefacts etc (just slow but that's printer hw limitation AFAIK).

No idea if BambuStudio accesses camera the same way (is it? port 6000, 0x40, 0x3000, 0x0 command - was looking for any docs for this API but no idea what it is) but if it is it rules out printer/firmware.

bambu123, 2023-12-04:
@arekm Can you help test the 1.8 beta version. 
[1.8.0 Public beta 2](https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62)
If that problem still exists, Please replace the BambuSource library with the following one in the plugin directory.
[libBambuSource.dylib.zip](https://github.com/bambulab/BambuStudio/files/13542635/libBambuSource.dylib.zip)

arekm, 2023-12-04:
I'm using 1.8.0.62 (beta2) on macos arm64 since its release two weeks ago.

The problem with "greenish" effect is no longer happening (also tested on intel machine). The "showing old frames problem" also seems to be gone/fixed.

The problem with starting the stream is still there. For example I'm trying to start it right now for over 9 minutes and it did not start. Testing this on 1.8.0.62  on two machines (one arm64 and one intel based) right now. While writing this comment print finished (10 minute print) ... and streams started showing up on both machines.

Also tested provided libBambuSource.dylib in that time but it didn't help - stream did not start until print finished.

Using p1p.py python test script I was able to start stream in about 3s few times in that 10 minute period, without a problem.

Now, when nothing is printing - stopping stream & starting it again works. Stream starts in ~2s from clicking "triangle". On both machines.

Doing second print test. Two BambuStudio running on both machines. On one did slicing and clicked "print". Stream stops on both machines (printer is downloading gcode). Then "Loading..." ... and I'm waiting.

p1p.py test - starts in ~3s.

Still waiting.... 4 minutes passed and it still did not load and it looks like it stopped trying (got "triangle" play button visible again and no "Loading..." message). Clicked play triangle and waiting. Closed one bambu studio and stop + play again on second. Waiting for 2 minutes with no results.

p1p.py test again - starts in 3s.

Print finished, stream didn't start. Stop & start - waiting. Stream this time started after about 2 minutes from finishing print and clicking stop/start.

One more time. Starting on arm64 machine with provided libBambuSource.dylib. In device tab stream is open and working. Preview -> Print plate. Studio switched to device tab, stream stopped, gcode fetched and now it is "Loading....". 10 minutes later, still "Loading..." and printer finishes print. Bambu Studio stopped trying "Loading" and now shows play triangle. Clicked... and stream started in 2-3s.

End of testing.

(https://status.bambulab.com/ shows all systems "Operational" if that matters)

bambu123, 2023-12-05:
Thank you for your detailed testing and handling the green screen issue. The startup issue is more like the printer not providing data. Your python script is connected to a LAN channel, while Studio uses a P2P channel, so it looks more like a P2P channel transmission issue.

bambu123, 2023-12-05:
@arekm Could you please provider the Studio Log And the printer log following these wiki.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

arekm, 2023-12-05:
Logs attached to support ticket US231205326001 but please keep discussion here (unless it needs to be private; then on ticket).

I see that beta 3 is released now but I did my testing few minutes earlier, on beta 2.

bambu123, 2023-12-05:
> Logs attached to support ticket US231205326001 but please keep discussion here (unless it needs to be private; then on ticket).
> 
> I see that beta 3 is released now but I did my testing few minutes earlier, on beta 2.
> 
> LAN channel & P2P channel... I have flaky LTE connection that breaks when there is a heavy rain (but yesterday/today weather was/is just fine, so that wasn't issue) and I remember Bambu Studio stream did not stop when rain started and "internet" stopped working. But don't remember if that was on X1C or on P1S printer.
> 
> Was LAN channel used before and now only P2P (which I assume is via some outside relay) is used? AKA now stream (assuming is started correctly and is visible) will stop if my internet breaks?

Thank you, we will analyze it immediately. Beta 3 did not change this issue, it doesn't matter.

bambu123, 2023-12-06:
> Logs attached to support ticket US231205326001 but please keep discussion here (unless it needs to be private; then on ticket).
>
> I see that beta 3 is released now but I did my testing few minutes [...]

Our printer engineer found some memory  usage abnormalities in the logs, but they are not sure what operation caused it. Insufficient memory can cause Liveview to fail. Therefore, please provide all printer logs, or attach them to the original ticket ticket.

arekm, 2023-12-06:
No corelogs for that day. 

````
lftp bblp@192.168...:~/corelogger> ls
-rw-rw-rw-   1 root  root     30512 Nov 22 20:09 corelogger-vr01.05.16.97_01P00A381000109_20231122200952_e.bin
-rw-rw-rw-   1 root  root     43664 Oct 08 09:41 corelogger-vr01.05.16.81_01P00A381000109_20231008094126_e.bin
````

And "recorder" is inaccessible over ftp :/

````
lftp bblp@192.168..:~> ls
drw-rw-rw-   1 root  root         0 Jan 01 1980 logger
drw-rw-rw-   1 root  root         0 Jan 01 1980 recorder
drw-rw-rw-   1 root  root         0 Sep 28 11:32 cam
drw-rw-rw-   1 root  root         0 Sep 28 12:30 cache
drw-rw-rw-   1 root  root         0 Sep 28 12:37 timelapse
drw-rw-rw-   1 root  root         0 Oct 05 14:31 corelogger
drw-rw-rw-   1 root  root         0 Aug 08 19:03 model
lftp bblp@192.168...:~> cd recorder
lftp bblp@192.168.68.46:~/recorder> ls
ls: Access failed: 550
lftp bblp@192.168...:~/recorder> debug 10
lftp bblp@192.168...:~/recorder> ls
---- CWD path to be sent is `~/recorder'
---> CWD recorder
<--- 550
ls: Access failed: 550
lftp bblp@192.168...:~/recorder>
````

will have to get physically to the printer to see if accessing sd card directly will allow me to see "recorder" logs. Will do that later.

(Tried Bambu Handy logs upload, too but that didn't seem to do the job: "01P00A381000109_all_20231204124604_20231205194600_enc.zip
Incomplete")

bambu123, 2023-12-06:
> No corelogs for that day.
>
> ```
> lftp bblp@192.168...:~/corelogger> ls
> -rw-rw-rw-   1 root  root     30512 Nov 22 20:09 corelogger-vr01.05.16.97_01P00A381000109_20231122200952_e.bin
> -rw-rw-rw-   1 [...]

They are in the 'logger' directory.

arekm, 2023-12-06:
All logs from logger directory for Dec 05 (close to my testing time) were attached to the ticket. You mean you want logs from other days/hours, too?

bambu123, 2023-12-06:
> All logs from logger directory for Dec 05 (close to my testing time) were attached to the ticket. You mean you want logs from other days/hours, too?

We need logs from the time the printer started up, including about 6 hours before the problem occurred.

arekm, 2023-12-06:
To make things simpler I've uploaded all logs I have in "logger" directory.

bambu123, 2023-12-07:
> To make things simpler I've attached to support ticket all logs I have in "logger" directory.

Thank you very much! We'll soon check the problem.

bambu123, 2023-12-07:
@arekm 
We have found a memory leak issue on the data channel, which can be resolved in the next version. Insufficient memory can cause Liveview to fail. Before the new version, it is recommended to restore by restarting the printer.

togeis, 2024-01-02:
But why this problem occurs only on MacOS and not in Windows (proven with a Windows10 FusionVM)? So why a printer restart should help ? Maybe a problem with the network plugin in MacOS. On my side, the problem occurs even when the printer is fresh restarted.

bambu123, 2024-01-03:
> > But why this problem occurs only on MacOS and not in Windows ...
> 
> I can confirm the problem has nothing to do with the printer. It's only MacOS and only on Intel macs. There are no video glitches on my iPhone, MacBook M1, Windows natives, Windows VM or even Linux (VM). **ONLY happens on my Intel iMac**.
> 
> In my case, the issue presents on an iMac with a **AMD Radeon Pro 5300 4 GB** GPU.

Does that have anything to do with the macOS version?

bambu123, 2024-01-04:
> > The trouble is that we did not find a stable environment for reproduction.
> 
> Understood. Perhaps then you have not even seen the visual effects in detail? I have posted a video for you at: https://youtu.be/PnZxwFpxurw.
> 
> > Does that have anything to do with the macOS version?
> 
> Both my MacBook M1 and iMac 27" Intel are running macOS Sonoma 14.2.1. There have been at least two general updates from the time this problem was first noticed. As far as I can recall, all macOS 14 but maybe also macOS 13 on Intel.

I just found out that your problem is different from others. Others have a green stripe problem and it's a P1 series printer.

bambu123, 2024-02-05:
@gruvin @arekm @Magalex2x14 @togeis 
Can anyone help test the following fix plugin?
How TO:
1、Replace libBambuSource.dylib with the following， it is located at ~/Library/Application\ Support/BambuStudio/plugins
2、Open libBambuSource.dylib by context menu, click OK on the warning dialog.
3、Open BambuStudio
4、Play liveview

[libBambuSource 2.dylib.zip](https://github.com/bambulab/BambuStudio/files/14160165/libBambuSource.2.dylib.zip)

arekm, 2024-02-05:
@bambu123  For me the problems were gone earlier with existing P1S firmware (1.5) and bambu studio  (1.8.4.51). Sadly can't tell what exactly fixed it (was it just printer restart (the memory leak thing that was found) or new firmware or new bambu studio).

But tested libBambuSource 2.dylib.zip anyway. Works fine (no green problem; no jumping problem; no long start delay problem; stopping view and starting it again takes about 2-3s and seem to work reliably). That's on macbook pro m3.

Magalex2x14, 2024-02-05:
Hello everyone! I'm currently on a business trip, and I won't be able to check it out for at least a week.

togeis, 2024-02-05:
Hi, the new lib is much better. See attached videos. old_lib.mp4 is stuttering and new_lib.mp4 is smooth, with a slight ghosting. I have a iMac Retina 5K 27" with Ventura 13.6.3.

https://github.com/bambulab/BambuStudio/assets/5475386/79a2972d-97cb-49b4-91ae-34412ccaa228

https://github.com/bambulab/BambuStudio/assets/5475386/e945edad-0933-44b4-99cb-9e612d37a76d

bambu123, 2024-02-06:
Resolved, please follow the prompts on Studio to update the plugin to 01.08.04.05.

closed 2024-02-06 by bambu123