bambulab/BambuStudio #2737: Eight-character HEX color codes

JeffreyBrew, 2023-10-16:
**Is your feature request related to a problem? Please describe.**
Currently, when selecting the customer color of a filament, only six characters are allowed for the HEX color code (RRGGBB).  The the current six-character HEX color code does not allow for a colorless transparent (aka, clear) filament.  A HEX color code of #000000 represents Black.

**Describe the solution you'd like**
Since printer filament is available in either opaque or transparent, Prusa slicer should be updated to accept eight-character HEX color codes (RRGGBBAA).  A HEX color code of #00000000 would represent a colorless transparent (aka, clear) filament.

**Describe how it would work**
The first 6 digits are interpreted identically to the current 6-digit notation. The last pair of digits, interpreted as a hexadecimal number, specifies the alpha channel of the color, where 00 represents a fully transparent color and ff represent a fully opaque color. 

**Describe alternatives you've considered**
None.

**Additional context**
See https://drafts.csswg.org/css-color/#hex-notation for more information.

nihilisticzebra, 2024-02-09:
> I appreciate your concern for the general user but, I think this will be a non-issue for general users. There was a time when RGB was new and the general users figured it out. RGBA is simply RGB with an Alpha channel indicating the degree of transparency. Transparency is nothing new. Every graphics program that offers transparency utilizes Alpha channels. Blender, GIMP, InkScape all offer Alpha channel (transparency). My guess is the general users already understand RGBA.

its irritating that Bambu hasn't fixed this yet on PC

JeffreyBrew, 2024-02-09:
> its irritating that Bambu hasn't fixed this yet on PC

Agreed.