bambulab/BambuStudio #2739: Error Building via Script for Ubuntu 22.04.3

CStolle4, 2023-10-16:
### Bambu Studio Version

NA

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

Bambu Lab P1S

### How to reproduce

1. Update CMake to latest version (cmake version 3.28.0-rc1)
2. Follow the Linux compile guide (https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)
3. On Step 4 (./BuildLinux.sh -dsi), the OCCT files are out of place, raising a fatal error.

### Actual results

-CMake Error at deps/build/dep_OCCT-prefix/src/dep_OCCT-build/OpenCASCADEConfig.cmake:95 (include):
  include could not find requested file:

    /home/XuserX/BambuLabs/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-build/OpenCASCADEFoundationClassesTargets.cmake
Call Stack (most recent call first):
  src/libslic3r/CMakeLists.txt:447 (find_package)

### Expected results

This step should generate an appimage without fatal errors.

### Project file & Debug log uploads

Unable to generate project or log files due to fatal error while building app image

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-10-16:
@CStolle4 
could you give a full log for your building?

CStolle4, 2023-10-16:
[log.txt](https://github.com/bambulab/BambuStudio/files/12922621/log.txt)

I found that by moving the 
/home/XuserX/BambuLabs/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-build/CMakeFiles/Export/72daae0aedbd85ec34659064f69a4d53

to the expected target

/home/XuserX/BambuLabs/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-build

resolves the initial set of errors but gives me some new errors related to the OCCT install. I've also included that log file.

[log2.txt](https://github.com/bambulab/BambuStudio/files/12922622/log2.txt)

lanewei120, 2023-10-17:
it seems you mixed building steps 1-3 and step-4

they are two different building ways
in fact, after you finished steps 1-3, the building have been finished
you can find the output are installed in /home/XuserX/BambuLabs/BambuStudio/install_dir

if you want to use step-4, it doesn't needs step 1-3
and you can remove the /home/XuserX/BambuLabs/BambuStudio/deps/build
and  /home/XuserX/BambuLabs/BambuStudio/build

then try step-4 again
the output appimage will be located at  /home/XuserX/BambuLabs/BambuStudio/build/BambuStudio_ubu64.AppImage after building

CStolle4, 2023-10-17:
I thought that might be the case, but it didn't generate an appimage at the end of step 3. Where can I get that from?

Thank you so much for your help!

lanewei120, 2023-10-17:
> I thought that might be the case, but it didn't generate an appimage at the end of step 3. Where can I get that from?
> 
> Thank you so much for your help!

step-3 only generate the binary of bambu-studio

lane.wei@glock18:~/bamboo_slicer/install_dir$ ls -l ~/bambu_master/bamboo_slicer/install_dir/
total 32
drwxr-xr-x  2 lane.wei domain_users 4096 Sep 18 17:36 bin
drwxr-xr-x  3 lane.wei domain_users 4096 Aug 18 20:21 include
-rw-r--r--  1 lane.wei domain_users 1036 Jul 22 18:39 LICENSE.txt
-rw-r--r--  1 lane.wei domain_users 8683 Jul 22 18:39 README.md
drwxr-xr-x 17 lane.wei domain_users 4096 Sep  9 19:16 resources
-rw-r--r--  1 lane.wei domain_users  296 Oct  7 09:21 result.json
lane.wei@glock18:~/bamboo_slicer/install_dir$ ls -l ~/bambu_master/bamboo_slicer/install_dir/bin/
total 105344
-rwxr-xr-x 1 lane.wei domain_users 107869856 Sep 18 17:36 bambu-studio

CStolle4, 2023-10-19:
Thank you for all of your feedback!

I want to provide some additional feedback on the installation process for ubuntu. I just did a fresh reinstall of my ubuntu OS and am reinstalling BambuStudio. When you run:
`sudo apt install cmake`
Ubuntu will only install cmake version 3.22.1, which is not sufficient for building BambuStudio (it requires the latest version of cmake, which is 3.27.1). I had to f[ollow the steps on this askubuntu page](https://askubuntu.com/questions/355565/how-do-i-install-the-latest-version-of-cmake-from-the-command-line) to install the latest version of cmake. This fixed my install issues. After reinstalling Ubuntu, correctly installing cmake, and going through this process, my BambuStudio worked as expected on my machine. 

Is there a way you could update the [Linux Compiler page](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide) to reflect these issues with cmake?

lanewei120, 2023-10-19:
> Thank you for all of your feedback!
> 
> I want to provide some additional feedback on the installation process for ubuntu. I just did a fresh reinstall of my ubuntu OS and am reinstalling BambuStudio. When you run: `sudo apt install cmake` (which is Step 1 of the compile) Ubuntu will only install cmake version 3.22.1, which is not sufficient for building BambuStudio (it requires the latest version of cmake, which is 3.27.7). I had to f[ollow the steps on this askubuntu page](https://askubuntu.com/questions/355565/how-do-i-install-the-latest-version-of-cmake-from-the-command-line) to install the latest version of cmake. This fixed my install issues. After reinstalling Ubuntu, correctly installing cmake, and going through this process, my BambuStudio worked as expected on my machine.
> 
> Is there a way you could update the [Linux Compiler page](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide) to reflect these issues with cmake?

yes, I can modify it on the web, 
and also you can submit a modify to the git directly by https://github.com/bambulab/BambuStudio.wiki.git

closed 2023-10-19 by CStolle4

lanewei120, 2023-10-20:
@CStolle4 

thanks for doing these~

could you also show us which module needs cmake 3.27.7?
I just checked our cmake files, and can not find it