bambulab/BambuStudio #2740: German words on calibration

FWeirich1, 2023-10-16:
Hi,
can you rename the German "Kalibration" to "Kalibrierung" ? Is a better pronouncing.
Likewise, the texts at the flow rate could look like this:

**Wann sollte die Flusskalibrierung verwendet werden**
Nach der Verwendung der dynamischen Flusskalibrierung, kann es immer noch zu Extrusionsprobleme kommen, wie zum Beispiel:
1. Überextrusion: Überschüssiges Material auf Ihrem gedruckten Objekt, das Flecken oder Pickel bildet, oder die Schichten erscheinen dicker als erwartet und nicht einheitlich.
2. Unterextrusion: Sehr dünne Schichten, schwache Füllfestigkeit oder Lücken in der obersten Schicht des Modells, auch beim langsamen Drucken.
3. Schlechte Oberflächenqualität: Die Oberfläche Ihrer Drucke wirkt rau oder uneben.
4. Schwache strukturelle Integrität: Drucke brechen leicht oder wirken nicht so robust, wie sie sein sollten.

**Auto-Kalibrierung**
Die automatische Kalibrierung der Durchflussrate nutzt die Mikro-Lidar-Technologie von Bambu Lab und misst direkt die Kalibrierungsmuster. Bitte beachten Sie jedoch, dass die Wirksamkeit und Genauigkeit dieser Methode bei bestimmten Materialtypen beeinträchtigt werden kann. Insbesondere Filamente, die transparent oder halbtransparent sind, glitzernde Partikel aufweisen oder eine hochreflektierende Oberfläche haben, sind für diese Kalibrierung nicht geeignet und können zu unerwünschten Ergebnissen führen.

Die Kalibrierungsergebnisse können von Kalibrierung zu Kalibrierung oder von Filament zu Filament variieren. Die Genauigkeit und Kompatibilität dieser Kalibrierung wird im Laufe der Zeit durch Firmware-Updates weiter verbessert.

**Achtung!**
Die Kalibrierung der Durchflussrate ist ein fortgeschrittener Prozess, dieser sollte nur von Personen durchgeführt werden, die den Zweck und die Auswirkungen der Kalibrierung genau verstehen. Falsche Anwendung kann zu minderwertigen Drucken oder Druckerschäden führen. Bitte stellen Sie sicher, dass Sie den Prozess gelesen und verstanden haben, bevor Sie ihn durchführen.

Best Regards
Fabrice

Schokobecher, 2023-12-07:
@QingZhangBambu is this still relevant? Are you looking for a Community PR for this?

StoneLiBambu, 2024-01-22:
[image]

i have update the translation item in our own translation platform, it will be release in next version.

Stone Li（旺仔）

closed 2024-01-22 by StoneLiBambu