bambulab/BambuStudio #2742: Camera very often not shown in Bambu Studio

fiing, 2023-10-16:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11. V10.0.22623 Build 22623

### Additional system information

2 Year old machine with Intel i5, 16Gb RAM etc.

### Printer

Bambu P1P

### How to reproduce

[image]
In 80% of time I'm not able to show camera in Bambu Studio (see screen dump). Not only happend in this version - has been that for few last versions.
All other connection and handling with printer works very well. Start new job, Pause, see status etc.
So for camera the Oraca Slicer always works so I use this only for using the camera.

### Actual results

Quite rare that there is a connection but even then it sometimes break. But only camera connection. Status and so on still visible.
It's not a huge problem as I just use Orca but it would be nicer not to start 2 programs.

### Expected results

I should see the camera all the time - like in Orca Slicer.

### Project file & Debug log uploads

[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/12913152/BambuStudio.zip)
[F_WihaHexagonKeysv10.zip](https://github.com/bambulab/BambuStudio/files/12913160/F_WihaHexagonKeysv10.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-10-17:
It is possible that the Access code does not match, please enter the correct Access code. Please note that rebinding the printer will cause the printer Access code to change and become invalid.

fiing, 2023-10-17:
Hi.
  

  
No the Access code has not been changed for months. And as mentioned file transfer and status works flewless. Only the camera thats unstable on Bambu Studio.
  
On Orca Slicer it works every time (same Access code)
  

  
Best regards
  
Finn

bambu123, 2023-10-17:
> Hi. No the Access code has not been changed for months. And as mentioned file transfer and status works flewless. Only the camera thats unstable on Bambu Studio. On Orca Slicer it works every time (same Access code) Best regards Finn

Have you tried re-entering the access code? If not, manually delete the user_access_code recorded in BambuStudio.conf.
[image]

Geekay1972, 2023-11-20:
I have exactly the same problem on my Macbook. What's interesting is that if I check with Bambu Handy everything is fine. The whole connection from Bambu Studio is quite crappy.

fiing, 2023-11-20:
I have started using Orca slicer as it works all the time + as you say the Bambu handy.

bambu123, 2023-11-21:
@fiing @Geekay1972 Please try the v1.8 version. [1.8.0 Public beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57)

childrentime, 2026-02-06:
@BambulabRobot why not close?