bambulab/BambuStudio #2744: Bambu Studio - Sprial Vase Mode - Producing "Seam/Artifact" on Lofted Surfaces at "Layer Change"

brandonraineri, 2023-10-16:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Enterprise

### Printer

Bambu Lab X1

### How to reproduce

1.  Import a Part with a lofted surface that will be printed in vase mode
2.  Slice the Part
3.  Find the "Layer Change" location on the lofted part in the preview and notices it creates a "seam" or position change.
5. This "Seam/Defect" shows up on the printed part

### Actual results

I've tried this on multiple different models with lofted parts.  Tried importing extreme quality STL, STEP, etc. and still produces this "seam/artifact".  I've also tried different extrusion widths, layer heights, etc.  It only shows up on the lofted surface and does not show up if the surface is simply vertical/extruded.

[image]

[image]

[image]

[image]

[image]

### Expected results

All walls should be smooth with no visible change in position

### Project file & Debug log uploads

[EGO Stubby Nozzle 70mm v1_ASA.zip](https://github.com/bambulab/BambuStudio/files/12915939/EGO.Stubby.Nozzle.70mm.v1_ASA.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-10-17:
There is a sharp change between this layer's end and the next layer's start., because of the z change

brandonraineri, 2023-10-17:
After doing more digging, it seems like this is a known bug in PS for some time now.  Seems like there are some scripts to post process the g-code but still somewhat experimental.

[https://github.com/prusa3d/PrusaSlicer/issues/2841](url)

GuzziRaz, 2023-10-26:
> There is a sharp change between this layer's end and the next layer's start., because of the z change

That sharp change **is** the bug. A sudden Z-hop defeats the whole idea with spiral/vase mode. There should be a tiny increase of Z all around the object, thus no seam.

Correct link for Prusa bug: https://github.com/prusa3d/PrusaSlicer/issues/2841

andrewboktor, 2023-11-29:
> > There is a sharp change between this layer's end and the next layer's start., because of the z change
> 
> That sharp change **is** the bug. A sudden Z-hop defeats the whole idea with spiral/vase mode. There should be a tiny increase of Z all around the object, thus no seam.
> 
> Correct link for Prusa bug: [prusa3d/PrusaSlicer#2841](https://github.com/prusa3d/PrusaSlicer/issues/2841)

This is not caused by a sudden change in Z, it's caused by a sudden change in XY since the layers don't stack perfectly on top of each other. So the nozzle moves horizontally from the end of one layer to the beginning of the next.

QuiNz0r, 2023-11-30:
Please I urge you to fix this, I mainly print vases and this bug is driving me insane

QuiNz0r, 2023-11-30:
> There is a sharp change between this layer's end and the next layer's start., because of the z change

That is the actual bug, there should be NO sharp Z change in Spiral vase mode!

GuzziRaz, 2023-12-01:
> This is not caused by a sudden change in Z, it's caused by a sudden change in XY since the layers don't stack perfectly on top of each other. So the nozzle moves horizontally from the end of one layer to the beginning of the next.

This is correct. Still a bug though - or rather, the current spiral mode is a hack. I understand it's not a trivial thing.

Very interesting reasoning, ideas and workaround (also hacks, but promising) in https://github.com/prusa3d/PrusaSlicer/issues/2841

Quoting myself [from there](https://github.com/prusa3d/PrusaSlicer/issues/2841#issuecomment-1832184456):
> > Cura fixed it for most parts ages ago. There is an option called smoot spiralized contour. From what I see it just interpolates the XY coordinates of each point of one spiral according to the one above. The first point isn't moved at all, the last point is 100% moved to where the first of the next spiral will be. If it's a perfect cylinder no point will move.
> 
> Assuming we slice from bottom up, can we not just do the opposite? Then we only need to know the layer below (which we do because it's already sliced): Interpolate the XY coordinates of each point of one spiral according to the one below. The **first** point is 100% moved to where the last of the previous spiral was, the **last** point isn't moved at all. If it's a perfect cylinder no point will move.

GuzziRaz, 2023-12-21:
Fixed in Orca - please merge!

QingZhangBambu, 2024-01-04:
it will be added in 1.9 version

m50434, 2024-02-27:
> it will be added in 1.9 version

At what time is the release of 1.9 ?
I really need this to print objects in spiral mode without z-seem!

Is there a 1.9 beta-download available?

Tomsko567, 2024-03-17:
Hi, I resolved this by accident. I increased the count of bottom layers and the artifacting I had went away after re-slicing. This has worked for me both times I have had this problem. The second time I used 6 bottom layers. I think it was artifacting on 3 layers.

andrewboktor, 2024-03-19:
> Hi, I resolved this by accident. I increased the count of bottom layers and the artifacting I had went away after re-slicing. This has worked for me both times I have had this problem. The second time I used 6 bottom layers. I think it was artifacting on 3 layers.

Is this comment relevant to this issue or was that posted by mistake? Can you elaborate it is indeed in the right place?

Tomsko567, 2024-03-20:
> Is this comment relevant to this issue or was that posted by mistake? Can you elaborate it is indeed in the right place?

Hi, this is the first time I have used a forum so apologies if I am not doing this right.

Yes, I found this page when I was having the same problem slicing in spiral mode when I was getting a  strange artifact/seam as described. When I used 3 base layers the artifact/seam was there. When I increased to 6 base layers the issue simply disappeared. Screen shots attached…

[image]
[image]

QingZhangBambu, 2024-04-22:
the new vase mode has been added, please update to latest version

closed 2024-04-22 by QingZhangBambu