bambulab/BambuStudio #2748: Single Object Multiple parts with Multiple Colours

sheminasalam, 2023-10-17:
Hi, I have made a number of 3D models with multiple parts in single 3mf file. I mostly use Fusion 360 for models and during modeling I assign colours to each part. But when I upload this 3mf file to the slicer, It assigns the primary filament colour to all the parts. I was wondering if the slicer would be able to distinguish the parts with different colours and assign either the original colour to the parts or even randon colours(which can be changed later). 

For example if my 3mf file has 5 colour parts, the slicer should be to detect them. Then it should find which colour has the maximum spread by area and assign the filament 1 to it. The second largest should be assigned the second filament and so on till the last. 

Will this be possible?

Also there should be an option to segregate parts according to colours on different filament beds and arrange them to avoid using transition towers for multi filament printing.

Morialkar, 2023-10-24:
If they are multi-parts when saving the 3mf file, do they show as multi-part in the slicer? If they do you should be able to set their colours independently in the slicer.

sheminasalam, 2023-10-24:
They show as multiple objects but the problem is that all the objects will be allocated for the first filament colour instead of the colours in which they were originally designed. We can set the colours independently but with about 100-150 parts its just not efficient. It is best for the slicer to have the ability to read the colours in which each is designed and assign filament colours accordingly.