bambulab/BambuStudio #2753: text height not zero

Tekniqueman, 2023-10-17:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

x1c

### How to reproduce

1 got text
2 select height and you can't select zero hight, even that the dept is set to more than zero

### Actual results

the text is 0.1 mm above surface.

### Expected results

I would like the height to be able to be 0.0 if depth is more than 0.0 that way you can add text at the bottom of the object (against the build plate ithout the need for support.

### Project file & Debug log uploads

[Body7.zip](https://github.com/bambulab/BambuStudio/files/12937649/Body7.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-10-18:
[image]
@Tekniqueman  In this case, you can change the text shape type to modifier.

Tekniqueman, 2023-10-18:
I’m sorry to say that I have tried several different text font and it won’t let me add text with a height of 0.00 mm. minimum is 0.1 mm.

Venlig hilsen
Villy Feltmann
Ejer

Tlf. +45 26137252
Gentoftegade 54 2.th. - 2820 Gentofte
***@***.***
***@***.***
***@***.***<https://www.facebook.com/ddv.nu>
***@***.***<https://twitter.com/DDV_NU>
***@***.***<https://www.linkedin.com/in/villyfeltmann/>
CVR 19613577

Tænk på miljøet, før du printer denne e-mail.
Think about the environment before printing this email.
I forbindelse med persondataloven af 25/5 2018 gør jeg opmærksom på, at Din Digitale Vicevært opbevarer følgende oplysninger om vores kunder/leverandører:

  *   Navn
  *   Adresse
  *   E-mail
  *   Telefonnummer
  *   CVR-nummer
  *   Relevante betalingsoplysninger
Hvis du som kunde fortsat ønsker, at modtage dine faktura via mail, skal du ikke foretage dig noget, og du giver os herved samtykke til, at opbevare ovenstående oplysninger
Ønsker du, at blive slettet fra vores kundekartotek, skal du venligst gøre opmærksom på dette via mail. Denne rettighed kan du til enhver tid benytte dig af.

[image]<https://user-images.githubusercontent.com/120697589/276046816-37e92833-4ebc-4c2f-9209-db8fb258a2e3.png>

zhimin-zeng-bambulab, 2023-10-18:
[image]

@Tekniqueman  When you set the text as a modifier, even if the text has a height, the result after slicing will not have a prominent part.

zhimin-zeng-bambulab, 2023-10-23:
@Tekniqueman  Do you still have any questions?

Tekniqueman, 2023-10-23:
As I wrote back then, I can’t get the thickness to be less than 0.10 and I have tried with a lot of different Fonts. So I’m still not happy 😉
***@***.***

Venlig hilsen
Villy Feltmann
Ejer

Tlf. +45 26137252
Gentoftegade 54 2.th. - 2820 Gentofte
***@***.***
***@***.***
***@***.***<https://www.facebook.com/ddv.nu>
***@***.***<https://twitter.com/DDV_NU>
***@***.***<https://www.linkedin.com/in/villyfeltmann/>
CVR 19613577

Tænk på miljøet, før du printer denne e-mail.
Think about the environment before printing this email.
I forbindelse med persondataloven af 25/5 2018 gør jeg opmærksom på, at Din Digitale Vicevært opbevarer følgende oplysninger om vores kunder/leverandører:

  *   Navn
  *   Adresse
  *   E-mail
  *   Telefonnummer
  *   CVR-nummer
  *   Relevante betalingsoplysninger
Hvis du som kunde fortsat ønsker, at modtage dine faktura via mail, skal du ikke foretage dig noget, og du giver os herved samtykke til, at opbevare ovenstående oplysninger
Ønsker du, at blive slettet fra vores kundekartotek, skal du venligst gøre opmærksom på dette via mail. Denne rettighed kan du til enhver tid benytte dig af.

zhimin-zeng-bambulab, 2023-10-24:
@Tekniqueman  Have you tried changing the text shape to modifier and then slicing?

Tekniqueman, 2023-10-24:
Ahh you mean the Move tool, yes that is possible to move it in level with the surface. I just think it should be possible to do the same directly in the text tool and even move it below surface.
***@***.***

Venlig hilsen
Villy Feltmann
Ejer

Tlf. +45 26137252
Gentoftegade 54 2.th. - 2820 Gentofte
***@***.***
***@***.***
***@***.***<https://www.facebook.com/ddv.nu>
***@***.***<https://twitter.com/DDV_NU>
***@***.***<https://www.linkedin.com/in/villyfeltmann/>
CVR 19613577

Tænk på miljøet, før du printer denne e-mail.
Think about the environment before printing this email.
I forbindelse med persondataloven af 25/5 2018 gør jeg opmærksom på, at Din Digitale Vicevært opbevarer følgende oplysninger om vores kunder/leverandører:

  *   Navn
  *   Adresse
  *   E-mail
  *   Telefonnummer
  *   CVR-nummer
  *   Relevante betalingsoplysninger
Hvis du som kunde fortsat ønsker, at modtage dine faktura via mail, skal du ikke foretage dig noget, og du giver os herved samtykke til, at opbevare ovenstående oplysninger
Ønsker du, at blive slettet fra vores kundekartotek, skal du venligst gøre opmærksom på dette via mail. Denne rettighed kan du til enhver tid benytte dig af.

zhimin-zeng-bambulab, 2023-10-25:
Not use Move tool. Only change the text_shape type. You can compare the results before and after modification.
The first picture shows the result of setting the text thickness to 2 and then slicing.
The second picture is the result of changing the type of text_shape to modifier and then slicing.
[image]
[image]

Tekniqueman, 2023-10-25:
Hi Zhimin Zeng,
Now that’s brilliant! Thank you for being patience with me and show me that Studio can do much more than meet the eyes, and my imagination. 😊

Best regards
Villy Feltmann
Ejer

Tlf. +45 26137252
Gentoftegade 54 2.th. - 2820 Gentofte
***@***.***
***@***.***
***@***.***<https://www.facebook.com/ddv.nu>
***@***.***<https://twitter.com/DDV_NU>
***@***.***<https://www.linkedin.com/in/villyfeltmann/>
CVR 19613577

Tænk på miljøet, før du printer denne e-mail.
Think about the environment before printing this email.
I forbindelse med persondataloven af 25/5 2018 gør jeg opmærksom på, at Din Digitale Vicevært opbevarer følgende oplysninger om vores kunder/leverandører:

  *   Navn
  *   Adresse
  *   E-mail
  *   Telefonnummer
  *   CVR-nummer
  *   Relevante betalingsoplysninger
Hvis du som kunde fortsat ønsker, at modtage dine faktura via mail, skal du ikke foretage dig noget, og du giver os herved samtykke til, at opbevare ovenstående oplysninger
Ønsker du, at blive slettet fra vores kundekartotek, skal du venligst gøre opmærksom på dette via mail. Denne rettighed kan du til enhver tid benytte dig af.

[image]<https://user-images.githubusercontent.com/120697589/277867658-680bbe20-a15a-4fac-b6bd-680e2b3a5296.png>
[image]<https://user-images.githubusercontent.com/120697589/277867705-011c6385-ae31-409b-ad0c-8ecacf9cab66.png>

closed 2023-10-25 by zhimin-zeng-bambulab