bambulab/BambuStudio #2754: A1 Mini travel problem

PowerChaos, 2023-10-18:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

intel I5 135400F
128GB Ram
Intel ARC 750

### Printer

A1 mlni

### How to reproduce

add any file you like
start slicing
enable show travel moves
it moves to the chute all the time on each layer like smooth timelapse does

### Actual results

it slows down the print a lot , for no reason 

### Expected results

just keep printing normal instead moving to the end  each single time

### Project file & Debug log uploads

choose any file you like , they are all the same
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-10-18:
https://wiki.bambulab.com/en/software/bambu-studio/Timelapse
@PowerChaos  Please check the wiki, which explains why these travels are needed. If you don’t want them, you can turn off time-lapse photography when sending for printing.

PowerChaos, 2023-10-18:
Thank you,
it also got explained to me why they are needed and it is indeed not a bug
i guess it is because the slicer says "smooth" vs "Traditional" and compared to the X1C it does something totally different on the A1 mini

Thank you

closed 2023-10-18 by PowerChaos