bambulab/BambuStudio #2756: Backspace not deleting other objects on subsequent backspaces

keithel, 2023-10-18:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 23.04 Lunar Lobster

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i7-13700H
Memory: 64GB
Graphics Card: RTX A1000 6GB Laptop GPU

### Printer

Bambu Lab P1P

### How to reproduce

1. Create or open a new project.
2. Add two objects. Any will do - I used Cube primitive.
3. Click to select one of the objects.
4. Press `<Backspace>` (Object will be deleted)
6. Click to select the other object.
7. Press `<Backspace>`

In order to be able to use `<Backspace>` to delete another object, you need to clear the selection entirely, then select an object.
Only then will it allow you to delete the object with `<Backspace>`.

### Actual results

First object is deleted.
Second object is not deleted.

### Expected results

Both objects are deleted.

### Project file & Debug log uploads

[Cube.zip](https://github.com/bambulab/BambuStudio/files/13032871/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DanBao-Bambu, 2024-01-22:
Sorry for the late reply. Currently, Studio does not support Backspace to delete objects. It supports Delete to delete objects. And there are two types of deletion positions, one is to select deletion in the 3D interface, and the other is to select deletion in the Object List. Is the situation you mentioned that the selection and deletion of the 3D interface cannot achieve continuous deletion?

keithel, 2024-01-22:
Ok - tested this out again.
Two things - that probably should be split into two separate tickets:

1. Add a negative part to something. Right click it and see that the context menu indicates that the keyboard shortcut for "Delete" is "Backspace". Click on the negative part again to dismiss the context menu. Press <Backspace>. Negative part remains.
2. Add two .stls to the build plate. Second part is selected. Press <Delete>, nothing happens. Select the first part you added, press <Delete>. The second part is deleted. Select the remaining part on the build plate. Press <Delete>. Nothing happens.

DanBao-Bambu, 2024-01-23:
I have performed some related operations on the latest version of Studio released, and the operation process is presented by gif. Please confirm if this operation is the problem you described.
[image]

In addition, my shortcut operations are all Del keys, and currently the Back key is not supported for deleting selected parts.

keithel, 2024-01-24:
Interesting. For part 1, in your case, the context menu properly reports the shortcut key is \<Del>, but in my case (Ubuntu AppImage, today it's version 1.8.4.51) it displays the shortcut key is \<Backspace>.  With negative part selected, and no context menu showing, neither \<Del> or \<Backspace> delete the object. The context menu's "Delete" entry will delete it, however.
[image]
In the screencast here, after dismissing the context menu, I press the \<Delete> and \<Backspace> keys - and nothing happens.
Same after I delete the negative part and the actual part is selected. I press \<Delete> here, nothing happens, and I have to delete it from the context menu.

Note: This does not happen every time. It seems to happen every other time I try to reproduce it(? that's the best I can characterize it, but I don't think that's quite right either).

Part 2 still happens for me, but I can characterize it a bit better, and I'll show a screencast:
[image]
Here's the steps I take in the screencast:

1. I load an object
2. Clone it once
3. *deselect the clone*
4. Reselect the clone
5. Press \<Delete> and \<Backspace> a few times (nothing happens)
6. Select the first object
7. Press \<Delete> (object gets deleted)
8. Select the clone (the remaining object left)
9. Press \<Delete> and \<Backspace> a few times (nothing happens)
10. Right click to bring up context menu
11. Select "Delete" menu item. (object is deleted)

Now, note, if you do not do step 3 (deselect the clone), subsequently pressing \<Delete> will actually delete the object. The remaining object, if selected, will not delete with the \<Delete> key. Now, if you select something else (I have some other objects on a different build plate), then select the remaining object on this build plate, \<Delete> will delete the object.

DanBao-Bambu, 2024-01-26:
Okay, thank you for providing me with more information. I will try to reproduce and fix it.