bambulab/BambuStudio #2761: 有关bambu lab的黑色渲染问题

littleball01, 2023-10-19:
**Is your feature request related to a problem? Please describe.**
[image]
黑色耗材在bambulab下完全看不清细节，非常麻烦

**Describe the solution you'd like**
改变一下渲染的参数是否可行？或者用深灰色代替黑色

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Haidiye00, 2023-10-19:
This issue will be resolved in V1.8 and will be released in the near future

arekm, 2023-11-09:
[image]

It's way better in 1.8 beta. Thanks!

Haidiye00, 2023-11-09:
Thank you for your use and trust. We will continue to work hard

closed 2023-11-09 by Haidiye00