bambulab/BambuStudio #2762: Extra colour layers

KobaRiley, 2023-10-19:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Printer

Bambu X1 Carbon

### How to reproduce

Open attached 3mf file, and colour the cental square a diffent colour from large square. Set layer thickness to 0.2mm. Slice.

### Actual results

The first layer is as expected, the second layer only uses one colour, and the 3rd layer again uses two colours. 
When slicing at other layer heights, there is always a minimum of three layers, even if the small square thickness is less than the layer height. This leads to unncecessary filament changes, wasting filament and time.
[image]
[image]
[image]

### Expected results

There should be a single multicolour layer, and then only one colour used afterwards in the case of the attached example.

### Project file & Debug log uploads

There is no error, so not including any log file. 
[Example.zip](https://github.com/bambulab/BambuStudio/files/13039961/Example.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-11-13:
Currently, Studio'rule is that the paint on the top and bottom surfaces will be embedded in the solid fill.
This model has 3 solid layers at the bottom, so the color is embedded in 3 layers. The second layer only displays one color, which is a BUG and is related to the geometric data of the model. We will fix this problem later.

zhimin-zeng-bambulab, 2024-01-17:
[image]
This model is composed of two geometric bodies, as the picture shows.
Studio'rule is that the paint on the top and bottom surfaces will be embedded in the solid fill. This model has 3 solid layers at the bottom, so the color is embedded in 3 layers. 
But if this layer is a surface (top or bottom), the color of the original surface will be displayed first. In this case, the second layer is the bottom surface, and its color is white, so white is displayed first.

zhimin-zeng-bambulab, 2024-01-17:
[image]
[image]

You can take the following steps to avoid this problem:
1. Split this model into parts
2. Change the filament of Example_2 to 2

zhimin-zeng-bambulab, 2024-01-24:
@KobaRiley  Does the above methods solve your problem?

zhimin-zeng-bambulab, 2024-02-04:
A solution was given, and the user did not respond for a long time. Close it.

closed 2024-02-04 by zhimin-zeng-bambulab