bambulab/BambuStudio #2764: Crashing when changing print order of objects

castingflame, 2023-10-19:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 - Version	10.0.19045 Build 19045

### Additional system information

System Manufacturer	Gigabyte Technology Co., Ltd.
System Model	Z270N-WIFI
Processor	Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz, 4201 Mhz, 4 Core(s), 8 Logical Processor(s)
BIOS Version/Date	American Megatrends Inc. F5, 06/07/2017
Total Physical Memory	31.9 GB
Page File Space	4.75 GB
1TB SSD

### Printer

X1C

### How to reproduce

Trying to reorder objects in the print order list crashes application.

### Actual results

100% closes/crashes the app.

### Expected results

To be able to change the order of prints when using 'Print by Object' not 'layer'.

[log.zip](https://github.com/bambulab/BambuStudio/files/13042397/log.zip)
[CNC Umbilical mounts.zip](https://github.com/bambulab/BambuStudio/files/13042405/CNC.Umbilical.mounts.zip)

### Project file & Debug log uploads

[CNC Umbilical mounts.zip](https://github.com/bambulab/BambuStudio/files/13042415/CNC.Umbilical.mounts.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/13042416/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

castingflame, 2023-10-19:
castingflame@gmail.com

LiZ-Li-BBL, 2023-11-09:
Hello,
After checking your 3mf project file, I noticed that you seem to want to print all model objects on one plate, but there are two objects that are out of the boundary. 
If try to drag & drop an object from "Outside" to "Plate", Bambu Studio will crash. We will fix the crash in this scenario in the future.
[image]

As a temporary solution, I sugget you to place all the model objects inside the plate, and the drag & drop function of objectlist can work well.