bambulab/BambuStudio #2765: Flow::spacing() produced negative spacing. Did you set some extrusion width too small?  Printing with multiple colors

Jamesfrompqs, 2023-10-19:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Built from source by yourself or others

### OS version

Windows 11 Home 22H2

### Additional system information

Device name	HP-LT
Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz   1.80 GHz
Installed RAM	32.0 GB
Device ID	B491DEF8-16BE-4BBA-B3A8-2989F470F5E4
Product ID	00325-81251-02850-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points

### Printer

Bambu Labs X1

### How to reproduce

1. import STL
2. Color painting {multiple colors)
3. Slice plate
4. Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

### Actual results

[image]

### Expected results

Successful slice  

### Project file & Debug log uploads

[Key chain.zip](https://github.com/bambulab/BambuStudio/files/13046286/Key.chain.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Jamesfrompqs, 2023-10-19:
Problem seemed to happen after updating to the new Bambu system.

QingZhangBambu, 2023-11-24:
@Jamesfrompqs the key chain.zip seems empty, could you upload a new one?

QingZhangBambu, 2025-05-30:
closing due to no response

closed 2025-05-30 by QingZhangBambu