bambulab/BambuStudio #2766: SIGTRAP in BambuStudio on Debian 11

nandbert2, 2023-10-19:
### Bambu Studio Version

Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage

### Where is the application from?

Bambu Lab github releases

### OS version

Debian GNU/Linux 11 (bullseye)

### Additional system information

Kernel: Linux 5.10.0-26-amd64
Architecture: x86-64

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. sudo apt-get update
2. sudo apt-get upgrade
3. ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage

### Actual results

BambuStudio crashes with: 

(bambu-studio:10296): Gdk-ERROR **: 21:18:44.246: The program 'bambu-studio' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadAccess (attempt to access private resource denied)'.
  (Details: serial 6835 error_code 10 request_code 152 (GLX) minor_code 26)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

[image]

### Expected results

Normal startup of gui

### Project file & Debug log uploads

[debug_Thu_Oct_19_20_56_02_8257.log.0.gz](https://github.com/bambulab/BambuStudio/files/13046793/debug_Thu_Oct_19_20_56_02_8257.log.0.gz)
[Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage.log.gz](https://github.com/bambulab/BambuStudio/files/13046796/Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage.log.gz)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-10-20:
@nandbert2 As it says, run it with the GDK_SYNCHRONIZE environment

set GDK_SYNCHRONIZE 1
gdb ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage

nandbert2, 2023-10-20:
here's the log with GDK_SYNCHRONIZE set to1: 

[Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage.debug.2023-10-20.log.gz](https://github.com/bambulab/BambuStudio/files/13055390/Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage.debug.2023-10-20.log.gz)

nandbert2, 2023-10-20:
Today these libraries were updated:

gir1.2-javascriptcoregtk-4.0 gir1.2-webkit2-4.0 libjavascriptcoregtk-4.0-18 libjavascriptcoregtk-4.0-dev libwebkit2gtk-4.0-37 libwebkit2gtk-4.0-dev

After upgrading, the crash is gone.
BambuStudio starts normal as expected.

closed 2023-10-20 by nandbert2