bambulab/BambuStudio #2767: eSun PLA+ = ABS

chuk219, 2023-10-20:
### Bambu Studio Version

1.7.6.92 and 1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel i9-13900H
Memory: 32GB 6000MHz LPDDR5
GPU: RTX4070 (mobile)

### Printer

Bambu Lab P1P

### How to reproduce

1. Go to the device tab
2. In the AMS filament selector area, select edit material
3. Select eSun PLA+ in the filament selector drop down
4. Close the edit material window
5. The box for the material will now be labeled ABS

### Actual results

The AMS filament window now shows ABS in the slot where eSun PLA+ was selected.

### Expected results

The AMS filament window used to and should show PLA+ in the slot where eSun PLA+ was selected.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13049699/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

joshkurtzman, 2023-10-20:
I am experiencing the same behavior with Generic PETG. No matter what I do it is showing as ABS once I save.  Bambu Studio Version 1.7.7.89 on MacOS 14.0

StoneLiBambulab, 2023-11-06:
There is a bug when getting the filament type, will be fixed in 1.8.xx

Thanks!
Stone Li

StoneLiBambu, 2024-01-17:
fixed.

closed 2024-01-17 by StoneLiBambu