bambulab/BambuStudio #2769: shrinkage of filament

riccalisci, 2023-10-20:
please add scale parameter in the filament setting preset (like orca slicer) for shrinkage compensation of some materials (ASA or ABS for exemple).
It would be really convenient and useful.

thanks

Gil80, 2023-10-22:
+1 on this request.
I don't understand why it is not there to begin with.

QingZhangBambu, 2023-10-24:
ok. I want to know why you want that.

riccalisci, 2023-10-24:
because I print a lot of PLA (it's more convenient  faster for the initial prototyping phase) and then in ASA/ABS, but I have to remember to increase and decrease the scale of the project every time I change material.  Furthermore, if for example you cut the object and the scale is at 100.5%, the scale resets to 100%, but the object remains at 100.5% of original dimension.

Gil80, 2023-10-24:
> ok. I want to know why you want that.

Each filament has different characteristics.
As you know, even color additives alter a filament's properties.

1. Having the "Shrinkage" field in the filament's properties tab, allows for a granular control over the filament in all the planes of X,Y,Z.
2. The **process** tab only applies shrinkage compensation on X and Y plane.
3. It's **not** reasonable to save the process tab per-filament when considering each filament has different shrinkage value.
4. It makes more sense to have a process per filament type (PLA, PETG, etc.), and the shrinkage per-filament-roll.
5. This feature is available on all common slicers except Bambu Studio. 
6. At the very least, have this field available when enabling the **Advanced** option.

Thank you.

QingZhangBambu, 2023-10-25:
ok, I will test it.

Gil80, 2023-11-30:
> ok, I will test it.

any idea if this feature request will make it into the next slicer release?

riccalisci, 2023-12-13:
Not yet

riccalisci, 2023-12-13:
> > ok, I will test it.
> 
> any idea if this feature request will make it into the next slicer release?

Not yet

tsmith35, 2023-12-30:
> > any idea if this feature request will make it into the next slicer release?
> 
> Not yet

This is already a feature within the filament parameters in OrcaSlicer. Perhaps the necessary code could be pulled from there?

[image]

stony2k, 2024-02-03:
I agree. My X1 prints PLA with exact 0.2 mm undersize. When scaling the part i. X and Y to 100.12 % the real part ist about 99,99% to the real measures, which it good enough. Some shrinkage option is much appreciated instead of scaling every part every time.

MarkJeronimus, 2024-02-10:
@stony2k If your shrinkage is is the same in mm for both large and small parts, then you have to use the "X-Y contour compensation" and/or "X-Y hole compensation" in the global options. Only if your shrinkage is measured in percent, then you will need the currently-missing "Shrinkage" parameter.

Gil80, 2024-02-10:
@MarkJeronimus that's just on the X-Y plane. The missing shrinkage option applies for the Z as well and it's much more manageable to set it per filament.

PanamonCreel, 2024-02-19:
@Gil80 Shrinkage compensations, like in Orca Slicer for example, only apply to the X-Y axis.  The Z axis usually is not so much effected by the material contraction after extrusion since the cooled layers below already shrank back when you stack a new one on top so there it is more of a matter of z axis movement/build platform position precision than material contraction.  Z axis shrink would maybe be a factor if you printed with very high build chamber temperatures and a material that has a high CLTE (e.g. unfilled Nylon,  PE,...etc.) however the compensation factor would still be less than that for the XY axis thus would need a separate entry.

riccalisci, 2024-04-03:
How is it possible that the shrinkage setting is not in the 1.9 beta yet? :(

marcosscriven, 2024-04-11:
Just adding support for this idea as well. Would be very handy - have been bitten many times by ASA parts that are 0.5% too small.

Ic33s, 2024-04-23:
THE feature which makes OrcaSlicer the only used Slicer used for my Bambu X1C. Don't undestand, that it not already integrated (along with x-y, x-z, y-z skew compensation - which is not doable otherwise in the closed source environment from Bambu Lab). I would love to have the new exact height slicing feature, but not on the cost of not having my shrinkage settings taken away. Without this feature the X1C is much worse than my old tuned in Prusa MK3s regarding dimensions, which is a shame.

QingZhangBambu, 2024-04-25:
this feature has been added to our plan list. but according to the schedules, it will be added later. sorry to keep you waiting

brandonraineri, 2024-05-14:
Please add this.  It’s a PITA scaling parts manually for say ABS/ASA.  Orca already has this implemented in the filament settings.

The original request is like 1.5yrs old now.

https://github.com/bambulab/BambuStudio/issues/350

emberprototypes, 2024-06-02:
Absolutely insane that this isn't a feature already. I've been doing this manually for years and can't believe how slow people have been to adopt this for accuracy sake 

Please add this very soon @QingZhangBambu

andredigenova, 2024-06-10:
+1 to this, pain manually scaling parts for each filament and then keeping track of what each material requires myself.

emberprototypes, 2024-06-10:
Orca slicer has this implemented and I'm considering changing to it for now until Bambu adds this in. Just a heads up for anyone else who needs this feature.

mtwomey, 2024-06-12:
+1 wanting this feature so I can switch back to Bambu slicer.

BigFansStudio, 2024-06-14:
I found that all the dimension of my printed parts is seriously inaccurate after the parts is cooling down.
For example,
I printed a rounded diffuser plate with initial diameter 60.5mm, after cooling it became 59.8mm.
I printed a long shaft with initial length 160mm, after cooling it became 159.4mm.
 
I just recevied my A1 two days ago and I am very disappointed with the XY accuracy. Even my 5 years-old Anycubic Mega X can print more accurate.
I am a Mechanical Engineer that always need to design parts and use it in my machine.
Bambulab printers provide a lots of feature and nice printing surface quality. However, without the XY calibration / Shrinkage compensation feature, Bambulab is only good for leisure and FUN purpose, such like printing toys, figures and stuffs that do not require a high accuracy.

j-collom, 2024-06-29:
@QingZhangBambu  Do you have any sort of estimated delivery date yet? Someone commented it is supposed to be in 1.9 is this correct? What is the estimated delivery date for 1.9?

QingZhangBambu, 2024-07-01:
@j-collom it is in the plan list of 1.10. So it will be released in the next public version( not the beta version )

Roobotics, 2024-09-20:
Really wishing this feature was already in there.. My prints look flawless, but I just found out they are also undersized.. I figured surely the software already has some generic shrinkage profiles based on the filament type or maybe even lazier, an average of the common ones, then users could tweak specifics later.. but no, it doesn't even exist as a possible option, despite being in prusa and orca. This is one of THE biggest issues anyone making critically sized parts is going to stumble into and waste a lot of filament in some cases. My 108.5mm parts came out to 107.9 - 108.08mm on Bambu. But came out flawlessly sized on my old SV06 just because running orca, all I needed was the very generic shrinkage value for the pure material, calibration of any sort wasn't even needed, it came out to 108.45-108.6 so had more variation(and didn't look as pretty either!), but was MUCH closer to the target value. Making us manually resize parts is a very bad practice, the software forgets how things were scaled between sessions I've also found out, impossible to keep track of. I see it's on the roadmap so that's good, but this is a huge sticking point a lot of folks have hit, and only a few will find this topic, and even less will bother to comment the need for it.. You are probably bleeding a lot of users over to orcaslicer and prusaslicer by missing something so inherent as material shrinkage, because a printer is all about manipulating materials. predating slicer softwares from before prusa/orca/bambu, also had basic XY compensations somewhere when needed. So not even having that as a fallback and forcing scaling isn't even a good fallback option, IMHO. 

TLDR: Thanks for all the hard work you do all the same, but please keep this as a priority, else this means something as basic as an XYZ cube is inaccurate at scale.. that's bad!

riccalisci, 2024-09-25:
@QingZhangBambu why do not added features yet? :(

QingZhangBambu, 2024-09-26:
it's in our plan. we tend to add it on beta2

tsmith35, 2024-10-17:
> unfortuanly not included in the beta 1.10.0.74 👎

Interestingly, this request will hit the one year mark in 3 more days. Happy birthday, Issue 2769. 🎂  😉

MartinChan3, 2024-10-30:
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81 1.10 beta2 has been released!And now you can shrink as you want.

closed 2024-11-04 by QingZhangBambu

j-collom, 2024-11-27:
@QingZhangBambu  - Thank you very much! Works great, sorry for my delay in responding. I know how frustrating it is when you work on code, check it in, then it take a long time before it makes it into a release. Thanks for all your hard work, Bambu makes a great product which you and your team are a big part of.