bambulab/BambuStudio #2772: Bambu Studio thinks PETG in AMS is PETG-CF

gfalgiano, 2023-10-22:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

WINDOWS 10

### Printer

X1 Carbon

### How to reproduce

Print a PETG model with Generic PETG loaded in AMS. Bambu studio won't let you select the PETG in the AMS as it thinks it is PETG-CF. This is confirmed in the Device tab. If you select Generic PETG, it changes to PETG-CF

### Actual results

[image]
[image]
[image]

### Expected results

I expect the ability to select an AMS slot with PETG loaded.

### Project file & Debug log uploads

[debug_Fri_Oct_20_22_07_18_20504.log.zip](https://github.com/bambulab/BambuStudio/files/13062243/debug_Fri_Oct_20_22_07_18_20504.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-10-23:
@sodachen22 @StoneLiBambulab please check this issue

bam-winterfell, 2023-10-23:
Hello @gfalgiano 
The user presets use the basic information of PETG, and the "Type" in "Basic Information" was modified to PETG-CF. This resulted in AMS being unable to distinguish between PETG and PETG-CF materials. The temporary solution is as follows:
1.Delete custom presets related to PETG-CF.
2.Modify the type corresponding to this preset
3.Version 1.8 will incorporate this fix

Wishing you a pleasant life.

StoneLiBambulab, 2023-11-06:
There is a bug when getting the filament type, will be fixed in 1.8.xx

Thanks!
Stone Li

closed 2023-12-21 by bam-winterfell