bambulab/BambuStudio #2775: Many filaments and 2 printer profiles are not synchronizing to cloud

FlowerOfTheSunPrints, 2023-10-22:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 22H2 Build 22621.2428

### Printer

Bambu Lab X1C and P1S

### How to reproduce

I installed Bambu Studio on a new computer today, logged into my Bambu account, and after filament + profile sync, I noticed that several of my filaments are missing.

I went back to my original computer and collected a log file which shows that my "new" filament profiles aren't getting a sync setting ID because "setting exceeds limit", "IOT_ERROR_RESOURCE_IS_OUT", with HTTP error code 400.
(Is there a limit to how many filaments I can sync?)

Also, it synchronized an older X1C printer profile, and didn't sync my custom P1S printer profile at all.

[image]

[image]

### Actual results

So in general, the problems so far are:
1. My P1S profile is entirely missing
2. The X1C profile is not my true "current" profile - it should say "w-offsets" at the end of the name
3. Many filament profiles are missing

### Expected results

Expected the profiles to sync.

### Project file & Debug log uploads

[Bambu Log-2023-10-22.log](https://github.com/bambulab/BambuStudio/files/13063744/Bambu.Log-2023-10-22.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

QingZhangBambu, 2023-10-23:
could you provide your user ID and user account? the concerned staff need these to check the issue。

bambu123, 2023-10-23:
Yes, there is a limit. The maximum number of presets for personal printers is 20. You can organize and delete some unused presets, or use the manual export and import function to synchronize more presets. If you really need to save more presets in the cloud, please make a request and we will study it.

FlowerOfTheSunPrints, 2023-10-23:
> could you provide your user ID and user account? the concerned staff need these to check the issue。

2840656513
flowerofthesunprints

FlowerOfTheSunPrints, 2023-10-23:
> Yes, there is a limit. The maximum number of presets for personal printers is 20. You can organize and delete some unused presets, or use the manual export and import function to synchronize more [...]

I would like to make that request. I do run a small business, what do I need to be eligible for a business account?

Schokobecher, 2023-12-24:
@bambu123 
Having the current limit (and your current usage, e.g., 17/20) represented in the slicer would be helpful to diagnose sync issues at a quick glance. Maybe even a pop-up once you save another profile once you are over the limit, informing the user of the implications.

bambu123, 2023-12-25:
> @bambu123 Having the current limit (and your current usage, e.g., 17/20) represented in the slicer would be helpful to diagnose sync issues at a quick glance. Maybe even a pop-up once you save another profile once you are over the limit, informing the user of the implications.

We have added this notification in v1.8. You can try it.

BCtopics, 2024-03-17:
> I would like to make that request. I do run a small business, what do I need to be eligible for a business account?

Did you ever figure this out? I'm running into a similar issue.